osrf/vrx #454: Dead light buoy and same dock placard symbols in scan-dock-deliver task

Tinker-Twins, 2022-04-12:
Whenever I launch the `scan_dock_deliver.launch` file, in the Gazebo world that is brought up (screenshot attached herewith), the `light_buoy` is dead (i.e. it is pitch black and does not display any color sequence) and all the dock placard symbols are exactly the same (`blue_circles`). It is worth mentioning however, that sometimes (extremely rare cases) the light buoy and the dock placard symbols work as expected. This is causing a lot of delay in testing our algorithms for the `scan_dock_deliver` task.

I am not sure if other team members or the developers and maintainers of this repository are experincing similar issue. Although, I found https://github.com/osrf/vrx/issues/163 similar to what I am experiencing.

[image]

Tinker-Twins, 2022-04-12:
Another weird thing that I recently noticed regarding the aforementioned issue is that sometimes the WAM-V camera is able to view color sequence of the light buoy even though the light buoy appears to be dead in Gazebo window.

The dock placard symbols, on the other hand, are visible to the WAM-V camera exactly the same as they appear in the Gazebo window.

caguero, 2022-04-12:
I think this should be fixed in https://github.com/osrf/vrx/pull/452. Could you confirm?

Tinker-Twins, 2022-04-12:
Yes @caguero, https://github.com/osrf/vrx/pull/452 does fix this issue.

closed 2022-04-12 by Tinker-Twins