osrf/vrx #457: Scoring for Perception Task

charlestmontagnoli, 2022-04-12:
Hello,

When my team runs the perception task we are correctly identifying which buoy is where (left to right) but with some localization inaccuracy. We expect to incur 2m of error per buoy, and in fact the debug message that the perception_task publishes to the command window shows that.
I believe based on how the scoring reads, that if we get an error of 2 for each buoy, our overall score should then be 2. However, the score slowly builds up over the course of the run to be somewhere around 2.888. I know that improving our localization accuracy will improve this score, I'm just a bit confused as to how this math breaks down for the scoring. I've attached an image below showing the debug message I'm talking about. Please let me know if we're misinterpreting the scoring calculation. Thanks.
[image]

M1chaelM, 2022-04-14:
Hi,
Your description of the scoring is correct. If you correctly identify every buoy your final score should be less than or equal to 2 meters. We're looking into this to make sure that is correct, and we're trying to reproduce the issue you describe. Are you seeing this behavior every time you run the perception task? Or is it just on some trial worlds?

M1chaelM, 2022-04-14:
@charlestmontagnoli  Would it be possible to post the complete output you're seeing leading up to this score? 

As far as I can tell, it looks like you are using a custom world, since the snippet you are showing doesn't match any of our practice worlds. Is that right? 

The score you are showing would correspond to identifying 8 objects correctly (but with a large localization error) and getting 1 wrong. If you have output showing that every object is reporting a score of 2 but the final score is not 2, then that is definitely a problem we need to fix asap.

M1chaelM, 2022-04-18:
Pasting from a 4/15/2022 email from @charlestmontagnoli:

> As for the issues I was seeing with the perception scoring, in testing we were noticing two things:
> Ocassionally it seemed as though the average score was incorrect
> In the example I sent on Github, all 3 buoys had an error of 2, but the score came out to be 2.8888. It is important to note that these were the first 3 buoys that ever appeared, so there were no previous scores that should be affecting the average.
> 
> The final buoys of the task are never scored properly
> In the testing environment we were given, the final set of 3 buoys end exactly at the end of the 300 second testing period. This seems to cause a problem where the guesses we send for this final set of buoys is never counted, as the task ends at exactly 300 seconds. I'm able to replicate this problem on my machine, and in all of my testing I have never seen myself earn any score for the final set of 3 buoys.
> I'm not sure what the best solution could be, but perhaps spawning this set of buoys in like 5 seconds earlier could fix it? So then there is no risk of the simulation ending before the guess is scored.

M1chaelM, 2022-04-18:
I think I have managed to reproduce this (with some variation). Below is output from running

```
roslaunch vrx_gazebo perception_task.launch
```
and waiting for the whole task to run through:
```
[Msg] PerceptionScoringPlugin::OnRunning
[Msg] PerceptionScoringPlugin: spawning red_0
[ INFO] [1650301259.659207788, 25.001000000]: New Attempt Balance: 1
[Msg] PerceptionScoringPlugin: spawning green_0
[ INFO] [1650301259.659271319, 25.001000000]: New Attempt Balance: 2
[Msg] PerceptionScoringPlugin: despawning red_0
[ INFO] [1650301264.687024458, 30.001000000]: New Attempt Balance: 1
[Msg] PerceptionScoringPlugin: despawning green_0
[ INFO] [1650301264.687089947, 30.001000000]: New Attempt Balance: 0
[Msg] PerceptionScoringPlugin: spawning black_0
[ INFO] [1650301269.724870426, 35.001000000]: New Attempt Balance: 1
[Msg] PerceptionScoringPlugin: despawning black_0
[ INFO] [1650301274.754649776, 40.001000000]: New Attempt Balance: 0
[ INFO] [1650301274.754825365, 40.001000000]: 
name: red_0
type: mb_marker_buoy_red
time: 5.000000
duration: 5.000000
error: 10.000000
[ INFO] [1650301274.754875068, 40.001000000]: 
name: green_0
type: mb_marker_buoy_green
time: 5.000000
duration: 5.000000
error: 10.000000
[ INFO] [1650301274.754911401, 40.001000000]: 
name: black_0
type: mb_marker_buoy_black
time: 15.000000
duration: 5.000000
error: 10.000000
[ INFO] [1650301274.754949318, 40.001000000]: 10.000000
[Msg] Object population restarted
[Msg] PerceptionScoringPlugin: spawning red_0
[ INFO] [1650301299.910397289, 65.002000000]: New Attempt Balance: 1
[Msg] PerceptionScoringPlugin: spawning green_0
[ INFO] [1650301299.910451446, 65.002000000]: New Attempt Balance: 2
[Msg] PerceptionScoringPlugin: despawning red_0
[ INFO] [1650301304.935220892, 70.002000000]: New Attempt Balance: 1
[Msg] PerceptionScoringPlugin: despawning green_0
[ INFO] [1650301304.935289565, 70.002000000]: New Attempt Balance: 0
[Msg] PerceptionScoringPlugin: spawning black_0
[ INFO] [1650301309.952964598, 75.002000000]: New Attempt Balance: 1
[Msg] PerceptionScoringPlugin: despawning black_0
[ INFO] [1650301314.971459401, 80.002000000]: New Attempt Balance: 0
[ INFO] [1650301314.971520284, 80.002000000]: 
name: red_0
type: mb_marker_buoy_red
time: 45.001000
duration: 5.000000
error: 10.000000
[ INFO] [1650301314.971557229, 80.002000000]: 
name: green_0
type: mb_marker_buoy_green
time: 45.001000
duration: 5.000000
error: 10.000000
[ INFO] [1650301314.971592962, 80.002000000]: 
name: black_0
type: mb_marker_buoy_black
time: 55.001000
duration: 5.000000
error: 10.000000
[ INFO] [1650301314.971621766, 80.002000000]: 13.333333
[Msg] Object population restarted
[Msg] PerceptionScoringPlugin: spawning red_0
[ INFO] [1650301380.408234853, 145.004000000]: New Attempt Balance: 1
[Msg] PerceptionScoringPlugin: spawning green_0
[ INFO] [1650301380.408299961, 145.004000000]: New Attempt Balance: 2
[Msg] PerceptionScoringPlugin: despawning red_0
[ INFO] [1650301385.435234939, 150.004000000]: New Attempt Balance: 1
[Msg] PerceptionScoringPlugin: despawning green_0
[ INFO] [1650301385.435313437, 150.004000000]: New Attempt Balance: 0
[Msg] PerceptionScoringPlugin: spawning black_0
[ INFO] [1650301390.482202236, 155.004000000]: New Attempt Balance: 1
[Msg] PerceptionScoringPlugin: despawning black_0
[ INFO] [1650301395.544089308, 160.004000000]: New Attempt Balance: 0
[ INFO] [1650301395.544138921, 160.004000000]: 
name: red_0
type: mb_marker_buoy_red
time: 125.003000
duration: 5.000000
error: 10.000000
[ INFO] [1650301395.544162576, 160.004000000]: 
name: green_0
type: mb_marker_buoy_green
time: 125.003000
duration: 5.000000
error: 10.000000
[ INFO] [1650301395.544202121, 160.004000000]: 
name: black_0
type: mb_marker_buoy_black
time: 135.003000
duration: 5.000000
error: 10.000000
[ INFO] [1650301395.544224107, 160.004000000]: 14.444444
[Msg] Object population restarted
[Msg] PerceptionScoringPlugin: spawning red_0
[ INFO] [1650301541.666942942, 305.008000000]: New Attempt Balance: 1
[Msg] PerceptionScoringPlugin: spawning green_0
[ INFO] [1650301541.666993134, 305.008000000]: New Attempt Balance: 2
[Msg] PerceptionScoringPlugin: despawning red_0
[ INFO] [1650301546.681411870, 310.008000000]: New Attempt Balance: 1
[Msg] PerceptionScoringPlugin: despawning green_0
[ INFO] [1650301546.681507648, 310.008000000]: New Attempt Balance: 0
[Msg] PerceptionScoringPlugin: spawning black_0
[ INFO] [1650301551.701756628, 315.008000000]: New Attempt Balance: 1
[Msg] 319.999000000  OnFinished
Segmentation fault (core dumped)
[gazebo-2] process has died [pid 2154469, exit code 139, cmd /opt/ros/noetic/lib/gazebo_ros/gzserver --verbose -e ode /home/mrmccarr@ern.nps.edu/vrx_ws/devel/share/vrx_gazebo/worlds/perception_task.world __name:=gazebo __log:=/home/mrmccarr@ern.nps.edu/.ros/log/0ddd8ddc-bf39-11ec-b17c-e19c186b6d43/gazebo-2.log].
log file: /home/mrmccarr@ern.nps.edu/.ros/log/0ddd8ddc-bf39-11ec-b17c-e19c186b6d43/gazebo-2*.log

```

M1chaelM, 2022-04-18:
I can also confirm the score shown in the `INFO` message after each loop is identical to the published score. From the output of `rostopic echo /vrx/task/info` the scores reported over the course of the demo are
```
remaining_time: 
  secs: 299
score: 0.0
---
remaining_time: 
  secs: 289
score: 20.0
---
remaining_time: 
  secs: 279
score: 10.0
---
remaining_time: 
  secs: 270
score: 10.0
---
remaining_time: 
  secs: 249
score: 30.0
---
remaining_time: 
  secs: 239
score: 13.333333333333334
---
remaining_time: 
  secs: 169
score: 33.333333333333336
---
remaining_time: 
  secs: 159
score: 14.444444444444445
---
elapsed_time: 
  secs: 293
score: 34.44444444444444
```

M1chaelM, 2022-04-18:
The problem seems to be caused by a special condition in the `perception_task.world.xacro`. We're setting a parameter that causes one trial to repeatedly loop until the world times out (see the `loop_forever` parameter):
```
    <plugin filename="libperception_scoring_plugin.so"
            name="perception_scoring_plugin">
      <!-- Parameters for ScoringPlugin base class -->
      <vehicle>wamv</vehicle>
      <task_name>perception</task_name>
      <initial_state_duration>10.0</initial_state_duration>
      <ready_state_duration>10.0</ready_state_duration>
      <running_state_duration>300</running_state_duration>
      <!-- Parameters for PopulationPlugin -->
      <loop_forever>true</loop_forever>
      <frame>wamv</frame>
      <!-- Pose of each object is expressed relative to the body frame
           of the object named in the frame field - i.e., relative to
           the wam-v-->
      <object_sequence>
        <object>
          <time>5</time>
          <type>mb_marker_buoy_red</type>
          <name>red_0</name>
          <pose>8 0 1 0 0 0</pose>
        </object>
        <object>
          <time>5</time>
          <type>mb_marker_buoy_green</type>
          <name>green_0</name>
          <pose>8 2 1 0 0 0</pose>
        </object>
        <object>
          <time>15.0</time>
          <type>mb_marker_buoy_black</type>
          <name>black_0</name>
          <pose>8 -2 1 0 0 0</pose>
        </object>
      </object_sequence>
    </plugin>
```
However, the plugin doesn't handle this case correctly. It just carries the score from the last loop over to the next loop. The good news is that we never do this in the actual competition worlds or practice worlds other than this demo. So it should not affect the scoring of the competition. We should still fix it though.

M1chaelM, 2022-04-18:
## Options:
1. Get rid of the `loop_forever` option and just fully specify the course. 
2. Reset the score to 0 between loops.
3. Make the scoring plugin smarter by tracking the total number of objects and total score separately, so we can correctly score multiple trials.

My favorite is option 1, since this parameter seems like much more trouble than it's worth for a demo. @caguero what do you think?

PR #467 (2022-07-27): Removed 'loop_forever' from perception scoring and perception task world
This PR removes the use of the `loop_forever` parameter in the scoring plugin for the perception task to resolve #457  (the course needs to be fully specified now). It also removes the line setting it to true in `perception_task.world.xacro`.
closed 2022-07-27 by rmvanarse