osrf/vrx #460: Duplicate Buoyancy Code and computing buoyancy twice

acxz, 2022-05-10:
It seems that buoyancy is calculated in two different places.
One method is located at [`/usv_gazebo_plugins/src/usv_gazebo_dynamics_plugin.cc#L291-L358`](https://github.com/osrf/vrx/blob/3408b8bbe3d794dbcdee14241dea57d9f8307f6d/usv_gazebo_plugins/src/usv_gazebo_dynamics_plugin.cc#L291-L358).

I believe this is run at every timestep as the USVDyanmicsPlugin is loaded in automatically (i think).

The other method is from [`/usv_gazebo_plugins/src/buoyancy_gazebo_plugin.cc`](https://github.com/osrf/vrx/blob/master/usv_gazebo_plugins/src/buoyancy_gazebo_plugin.cc).

This method seems to be optional and will be loaded in if the sdf file contains a block for the BuoyancyPlugin like shown in [`/vrx_gazebo/models/blue_projectile/model.sdf`](https://github.com/osrf/vrx/blob/ae1823d410595238c7223adc2d5516134d5abd34/vrx_gazebo/models/blue_projectile/model.sdf#L38-L53)

Since it seems like USVDyanmicsPlugin cannot be unloaded, does that mean that if I use the BuoyancyPlugin, buoyancy forces will be computed twice?

Should the BuoyancyPlugin be consolidated into the method in USVDyanmicsPlugin?

M1chaelM, 2022-06-21:
@acxz You are correct that buoyancy is calculated in two different locations, but this should not cause any duplicate calculation. Currently the `usv_gazebo_dynamics_plugin` calculates buoyancy for the USV only. The `buoyancy_gazebo_plugin` calculates the buoyancy for all other objects.

We may consider reorganizing so all buoyancy forces are calculated in the same place, but the main reason to do this is to improve readability of the code. If we do that, it's more likely that we would move the USV buoyancy calculation into the BouyancyPlugin than the other way around, since not all the objects that require this plugin are USVs.

acxz, 2022-06-21:
> move the USV buoyancy calculation into the BouyancyPlugin

This would be much appreciated to maintain readability.

M1chaelM, 2022-06-22:
Ok, sounds good. Since the code is not duplicated or computed twice I will close this issue. Feel free to open another if you want to suggest an addition to the documentation.

closed 2022-06-22 by M1chaelM

acxz, 2022-06-22:
I was hoping this issue could be kept open until the BuoyancyPlugin is consolidated.