osrf/vrx #464: when  I  run   roslaunch vrx_gazebo vrx.launch   get an error

huangzizhen, 2022-06-13:
```
Error: Non-unique names detected in type plugin in
<model name='short_navigation_course_0'>
  <frame name='red_bound_0::__model__' attached_to='red_bound_0::link'>
    <pose relative_to='__model__'>0 0 0 0 -0 0</pose>
  </frame>
  <link name='red_bound_0::link'>
...
<plugin name='BuoyancyPlugin' filename='libbuoyancy_gazebo_plugin.so'>
    <wave_model>ocean_waves</wave_model>
    <fluid_density>1000</fluid_density>
    <fluid_level>0.0</fluid_level>
    <linear_drag>25.0</linear_drag>
    <angular_drag>2.0</angular_drag>
    <buoyancy name='buoyancy_sphere'>
      <link_name>red_bound_0::link</link_name>
      <pose>0 0 -0.12 0 0 0</pose>
      <geometry>
        <sphere>
          <radius>0.325</radius>
        </sphere>
      </geometry>
    </buoyancy>
  </plugin>
```
but i can not find this plugin in folder.  could  someone help me,  thanks !!!

huangzizhen, 2022-06-14:
when i run this ```roslaunch vrx_gazebo vrx.launch ``` Gazebo can only show the ocean, not the coastline

rmvanarse, 2022-06-22:
We were unable to reproduce this issue. The buoyancy plugin can be found in `vrx/usv_gazebo_plugins/src`.
Can you make sure that you have followed all the installation instructions and have sourced `vrx_ws/devel/setup.bash` in the terminal that you're using?

rmvanarse, 2022-07-03:
Since we could not reproduce this issue, we are closing it for now. Feel free to reopen this if you face any additional difficulty.

closed 2022-07-03 by rmvanarse