osrf/vrx #465: Replacing the wamv model

jasonnguyenn, 2022-06-29:
Hello,
I'm new to the vrx environment and have been playing around with the codes in order to replace the wamv model with a model of my own. If I wanted to only replace the model with a mode of my own, how would I go about doing that? In addition, as far as the physics of the model goes, how would I manipulate the codes so that the model with present the 6 degrees of freedom (equations wise). If someone could point me in the right direction, that'll be great.

M1chaelM, 2022-07-15:
Hi! Have you had a chance to look through the [tutorials](https://github.com/osrf/vrx/wiki/tutorials) on our wiki? The tutorials for [customizing VRX](https://github.com/osrf/vrx/wiki/customizing_vrx_tutorials), in particular, might be a good place to get started.

M1chaelM, 2022-07-20:
Issue #476 may also be relevant. It seems like we could use a step-by-step guide for adding a custom vehicle. I've modified the title of this issue to reflect that requirment.

caguero, 2025-05-08:
The "Maritime" section of the [Gazebo Sim tutorials](https://gazebosim.org/api/sim/9/tutorials.html) offers a step-by-step guide that could be useful. Additionally, we created a [RoboBoat set of tutorials](https://github.com/osrf/vrx/wiki/tutorials#roboboat-overview) that could be used as an example of a different vehicle. Hopefully this is sufficient to your needs.

closed 2025-05-08 by caguero