osrf/vrx #471: [wave_gazebo_plugins] uses features of c++17

acxz, 2022-07-13:
Portion of build log:
```
In file included from /usr/include/gazebo-11/gazebo/physics/physics.hh:9:
/usr/include/gazebo-11/gazebo/physics/Collision.hh:205:28: error: ‘optional’ in namespace ‘std’ does not name a template type
  205 |       public: virtual std::optional<sdf::SemanticPose> SDFSemanticPose()
      |                            ^~~~~~~~
/usr/include/gazebo-11/gazebo/physics/Collision.hh:205:23: note: ‘std::optional’ is only available from C++17 onwards
  205 |       public: virtual std::optional<sdf::SemanticPose> SDFSemanticPose()
      |                       ^~~
In file included from /usr/include/gazebo-11/gazebo/physics/physics.hh:26:
/usr/include/gazebo-11/gazebo/physics/Light.hh:78:20: error: ‘optional’ in namespace ‘std’ does not name a template type
   78 |       public: std::optional<sdf::SemanticPose> SDFSemanticPose() const override;
      |                    ^~~~~~~~
/usr/include/gazebo-11/gazebo/physics/Light.hh:78:15: note: ‘std::optional’ is only available from C++17 onwards
   78 |       public: std::optional<sdf::SemanticPose> SDFSemanticPose() const override;
      |               ^~~
In file included from /usr/include/gazebo-11/gazebo/physics/physics.hh:28:
/usr/include/gazebo-11/gazebo/physics/Link.hh:617:20: error: ‘optional’ in namespace ‘std’ does not name a template type
  617 |       public: std::optional<sdf::SemanticPose> SDFSemanticPose() const override;
      |                    ^~~~~~~~
/usr/include/gazebo-11/gazebo/physics/Link.hh:617:15: note: ‘std::optional’ is only available from C++17 onwards
  617 |       public: std::optional<sdf::SemanticPose> SDFSemanticPose() const override;
      |               ^~~
In file included from /usr/include/gazebo-11/gazebo/physics/physics.hh:36:
/usr/include/gazebo-11/gazebo/physics/PhysicsEngine.hh: In static member function ‘static T gazebo::physics::PhysicsEngine::any_cast(const boost::any&)’:
/usr/include/gazebo-11/gazebo/physics/PhysicsEngine.hh:274:45: error: ‘any’ is not a member of ‘std’
  274 |           auto value = boost::any_cast<std::any>(_value);
      |                                             ^~~
/usr/include/gazebo-11/gazebo/physics/PhysicsEngine.hh:274:45: note: ‘std::any’ is only available from C++17 onwards
/usr/include/gazebo-11/gazebo/physics/PhysicsEngine.hh:274:45: error: ‘any’ is not a member of ‘std’
/usr/include/gazebo-11/gazebo/physics/PhysicsEngine.hh:274:45: note: ‘std::any’ is only available from C++17 onwards
/usr/include/gazebo-11/gazebo/physics/PhysicsEngine.hh:275:23: error: ‘any_cast’ is not a member of ‘std’
  275 |           return std::any_cast<T>(value);
      |                       ^~~~~~~~
/usr/include/gazebo-11/gazebo/physics/PhysicsEngine.hh:275:23: note: ‘std::any_cast’ is only available from C++17 onwards
/usr/include/gazebo-11/gazebo/physics/PhysicsEngine.hh:275:33: error: expected primary-expression before ‘>’ token
  275 |           return std::any_cast<T>(value);
      |                                 ^
make[2]: *** [CMakeFiles/Hydrodynamics.dir/build.make:132: CMakeFiles/Hydrodynamics.dir/src/Utilities.cc.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [CMakeFiles/Hydrodynamics.dir/build.make:146: CMakeFiles/Hydrodynamics.dir/src/Wavefield.cc.o] Error 1
make[2]: *** [CMakeFiles/Hydrodynamics.dir/build.make:160: CMakeFiles/Hydrodynamics.dir/src/WavefieldEntity.cc.o] Error 1
In file included from /usr/include/ignition/transport8/ignition/transport/Node.hh:42,
                 from /usr/include/gazebo-11/gazebo/rendering/Projector.hh:26,
                 from /usr/include/gazebo-11/gazebo/rendering/rendering.hh:34,
                 from /home/acxz/vcs/git/github/nswc-pcd-heron-stochastic-control/vrx_ws/src/vrx/wave_gazebo_plugins/src/Gazebo.cc:25:
/usr/include/ignition/transport8/ignition/transport/NodeShared.hh:298: error: ‘optional’ in namespace ‘std’ does not name a template type
  298 |       public: std::optional<TopicStatistics> TopicStats(
      | 
/usr/include/ignition/transport8/ignition/transport/NodeShared.hh:298: note: ‘std::optional’ is only available from C++17 onwards
/usr/include/ignition/transport8/ignition/transport/Node.hh:740: error: ‘optional’ in namespace ‘std’ does not name a template type
  740 |       public: std::optional<TopicStatistics> TopicStats(
      | 
/usr/include/ignition/transport8/ignition/transport/Node.hh:740: note: ‘std::optional’ is only available from C++17 onwards
make[2]: *** [CMakeFiles/Hydrodynamics.dir/build.make:76: CMakeFiles/Hydrodynamics.dir/src/Gazebo.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:2103: CMakeFiles/Hydrodynamics.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
cd /home/acxz/vcs/git/github/nswc-pcd-heron-stochastic-control/vrx_ws/build/wave_gazebo_plugins; catkin build --get-env wave_gazebo_plugins | catkin env -si  /usr/bin/make --jobserver-auth=3,4; cd -
```

M1chaelM, 2022-07-15:
Hi, thanks for submitting an issue! In addition to the output you've posted, would you mind writing up a short narrative summary? It's helpful to include:
* The behavior you expect to see.
* The behavior you are getting.
* Steps to reproduce.

Thank you!

acxz, 2022-07-15:
Sure!

> The behavior you expect to see.

Successful compilation

> The behavior you are getting.

Unsuccessful compilation

> Steps to reproduce.

`catkin build vrx_gazebo`

Let me know if you need more information. Compiler is gcc 12.1.0.

M1chaelM, 2022-07-18:
Thanks. In my opinion, the key to reproducing this issue is to update gcc to a version that will give an error instead of a warning. Running `catkin build vrx_gazebo` will not produce this output if you've followed our installation instructions and are using a fresh installation of Ubuntu 20.04.

Consequently, I would describe this issue as follows:

* Currently, some plugins require features of C++17, but flags set in `CMakeLists.txt` for `usv_gazebo_plugins` and `wave_gazebo_plugins` indicate C++17 is not required.
* This causes compilation to fail when using the latest version of gcc (e.g. 12.1.0).

Of course, it's totally reasonable to fix this, but slightly less urgent than it would be if compilation failed when following our installation instructions.

Are you using Ubuntu 22 or did you install gcc 12 from source?

acxz, 2022-07-18:
My OS is Archlinux, where gcc 12 is available in the package repos.

M1chaelM, 2022-07-18:
OK, thanks. This is not supported, so in the future please be sure to note in any issues that you are using a non-standard set up that we may not be able to reproduce. If you'd like to use a supported setup, you can follow the installation instructions available on our Wiki: https://github.com/osrf/vrx/wiki/system_setup_tutorials

Of course, since you have already forked the repository, there's no problem with adding support for whatever configuration you want to your fork.

Again, this is not to say that we can't resolve this particular issue, which looks very reasonable. But since you have been relatively active lately it seems like a good idea to get on the same page.

caguero, 2025-06-20:
Solved in https://github.com/osrf/vrx/pull/828.

closed 2025-06-20 by caguero