osrf/vrx #487: Improve support for custom vehicles

M1chaelM, 2022-07-22:
Currently we support the WAMV and CORA platforms. We have seen several requests for guidance on adding new vehicles. See:
* #434
* #465
* #476 
* #472 (discussion)

This issue could be partially resolved with a detailed tutorial, as suggested in #465. However, some aspects of our vehicle model are currently hard-coded to match the WAMV design (e.g. two cylindrical hulls) and even our mono-hull CORA vehicle is using this. In addition, some vehicles may require extending the thruster plugin's capabilities, as proposed in the following PRs:
* #479 
* #480 
* #481 

We should support these features if possible, but we need to be mindful of any potential impacts on VRX competitors--in particular, it would be good to double check that any modifications which will propagate back to the WAMV are faithful to our theory of operations and don't have any unintended effects for the competition, or at least can be disabled if they do.

acxz, 2022-07-23:
Thanks for putting up this list @M1chaelM.

Can you also add the following PRs which increase/correct the fidelity of the simulator?

- https://github.com/osrf/vrx/pull/427
- https://github.com/osrf/vrx/pull/466
- https://github.com/osrf/vrx/pull/485

acxz, 2022-07-23:
> support these features if possible

I think this is evident based on the PRs submitted.

> be mindful of any potential impacts on VRX competitors

For sure

> double check that any modifications which will propagate back to the WAMV are faithful to our theory of operations

I would be willing to get on call with you to discuss the math behind the PRs

> don't have any unintended effects for the competition

All PRs (Besides #427, #485, & #489) retain default values that do not change any behavior. The mentioned PRs fix the correct bearing, (if VRX competitors are using this quantity they would need to add a negative sign to maintain the original behavior in their code), and increase the accuracy in the buoyancy calculation. From the PRs the buoyancy has increased in simulation by about 100N (increase of only about 4%), but the visible behavior change of the vessel is not noticeably at all.

> at least can be disabled if they do

Many changes are in the plugins and those portions can be selectively not loaded.

The above two concerns are specific to each feature and the relevant conversation and concerns should be had on the respective PRs/issues for those.

caguero, 2025-10-30:
I'd also look at the [RoboBoat resources](https://github.com/osrf/vrx/wiki/tutorials#roboboat), as it's another example of using VRX with a different vehicle.

closed 2025-10-30 by caguero