osrf/vrx #507: How to change wave parameters?

bsb808, 2022-09-01:
## Scenario

Working on PR#500 and would like to test under wave conditions other then the default.

Consider the simple Sydney Regatta example: https://github.com/osrf/vrx/wiki/2023-running_vrx#the-sydney-regatta-environment

* How would one change the wave parameters for a higher seastate?

## What I tried

In the past we would change the `period` and `gain` parameters in the `ocean_waves` macro to change seastate - see https://github.com/osrf/vrx/wiki/changing_plugin_params_tutorial

Grepping for those SDF parameters I find that the `gain` parameter is set in many different places.
```
./vrx_ign/worlds/sydney_regatta.sdf:            <gain>0.3</gain>
./vrx_ign/worlds/sydney_regatta.sdf:            <gain>0.3</gain>
./vrx_ign/worlds/sydney_regatta.sdf:            <gain>0.3</gain>
./vrx_ign/worlds/sydney_regatta.sdf:            <gain>0.3</gain>
./vrx_ign/worlds/sydney_regatta.sdf:            <gain>0.3</gain>
./vrx_ign/worlds/sydney_regatta.sdf:            <gain>0.3</gain>
./vrx_ign/models/wam-v/model.sdf.erb:        <gain>0.3</gain>
./vrx_ign/models/wam-v_2/model.sdf.erb:        <gain>0.3</gain>
./vrx_ign/models/wam-v_2/model.sdf.erb:        <gain>0.3</gain>
./vrx_ign/models/coast_waves/model.sdf:              <gain>0.3</gain>
```

I changed all of the `gain` and `period` SDF tags to 1.0 and 10.0 respectively.  No change in the visual behavior of the wave field or the vessel motion.

caguero, 2022-09-08:
What you did is the right way to do it right now. You need to change it in all of those places. This will change when we register the parameters from a single place into a component. Then, all the other plugins will be able to read it using the component and the ECM.

bsb808, 2022-09-08:
Roger - I'll leave the issue open for now so that we have a reminder to address when the time is right.
Thank you.

M1chaelM, 2023-08-02:
This is now resolved. Here's a summary of what we did to fix it:
* We changed the wavefield plugin so it receives parameters from a single publisher, specified in the world.
  * #598  
* We added default values to the plugin so the user only needs to specify the ones we're actually using.
  * #678
* We updated the plugin so it is uses the same variables / format that we describe in our paper.
  * #651 
  * #709
* We wrote a [new tutorial explaining how to adjust the waves](https://github.com/osrf/vrx/wiki/wave_params_tutorial).

closed 2023-08-02 by M1chaelM