osrf/vrx #509: Alternative wave model example

srmainwaring, 2022-09-09:
@bsb808, @caguero and @j-herman, as a follow up from our call about wave models I've created a demo using the sydney_regatta world. I forked the gazebosim branch and swapped the wave models out for the ones in [asv_wave_sim](https://github.com/srmainwaring/asv_wave_sim).

https://user-images.githubusercontent.com/24916364/189438132-ccc66f38-fa33-42eb-8257-58629201d1d4.mov

The example uses the non-PBS shader version of the wave renderer. The ocean is a 21 x 21 grid of tiles, each 100m x 100m containing 256 x 256 cells for the FFT wave generation. 

## Changes

There are a few modifications:

1. The [asv_wave_sim](https://github.com/srmainwaring/asv_wave_sim) plugins build against Gazebo Garden (gz-sim7) which is the first version to have full support for macOS / Metal. It was going to be a lot of work to manage the namespace migration and retain backwards compatibility - so I opted to support Garden only.
2. The first commit on https://github.com/srmainwaring/vrx/commit/1ccfa9fcd7d20de400a6a4eb617b55e2eb11be5b deals with the upgrade to Garden.
3. The second commit https://github.com/srmainwaring/vrx/commit/e12c71fcbaad5117a45b07bbaeae6dd5ad2dcbe0 is a workaround for my installation, for some reason Gazebo will not load models downloaded by fuel.
4. The third commit https://github.com/srmainwaring/vrx/commit/3a8286af2386839dad93108f8738623391233ebe swaps the wave models over and adds the control to the GUI. There's not that much to it once the preparation has been done.

## Set up and run

1. Build osrf/vrx with the above changes.
2. Follow the instructions to install [asv_wave_sim](https://github.com/srmainwaring/asv_wave_sim). The master branch should be suitable. The GUI plugin must be manually built in `~/waves_ws/src/asv_wave_sim/gz-waves/src/gui/plugins/waves_control/build` as it's not fully integrated into the top level build files.
3. Set up the environment variables. I'm using something like the following:

```zsh
export GZ_SIM_RESOURCE_PATH=\
$HOME/.gz/fuel/models:\
$HOME/waves_ws/src/asv_wave_sim/gz-waves-models/models:\
$HOME/waves_ws/src/asv_wave_sim/gz-waves-models/world_models:\
$HOME/waves_ws/src/asv_wave_sim/gz-waves-models/worlds:\
$HOME/vrx_ws/install/share/vrx_ign/models:\
$HOME/vrx_ws/install/share/vrx_ign/worlds

export GZ_SIM_SYSTEM_PLUGIN_PATH=\
$HOME/waves_ws/install/lib:\
$HOME/vrx_ws/install/lib

export GZ_GUI_PLUGIN_PATH=\
$HOME/waves_ws/src/asv_wave_sim/gz-waves/src/gui/plugins/waves_control/build
``` 

Then run the world directly:

```zsh
gz sim -v4 -r sydney_regatta.sdf
```

mvanlobensels, 2024-02-02:
@srmainwaring Are you still able to control the WAM-V this way? The SimpleHydrodynamics also seems to compute the ship dynamics.

srmainwaring, 2024-02-03:
> Are you still able to control the WAM-V using the thrusters this way? The SimpleHydrodynamics also seems to compute the ship dynamics.

If the `gz-sim` thruster plugin is used, control is independent of the model for waves and the choice of hydrodynamics.

The example above uses plugins from asv_wave_sim to control the wave physics and visuals and another for the hydrodynamics. These plugins include non-linear restoring forces not captured in SimpleHydrodynamics, which uses a model that assumes small displacements from a mean position.

srmainwaring, 2024-02-08:
Closing - no further action required at this stage.

closed 2024-02-08 by srmainwaring