osrf/vrx #534: image_rawleul opencvlo boneun bangbeob 24 / 5,000 번역 결과 How to view image_raw with opencv

mazik1496, 2022-10-25:
Using the vrx open source example, we are going to output the image that comes out through the image_raw topic to opencv. I implemented the code, but only a black screen is output and no proper video is displayed. I want to ask how to solve this.

`#!/usr/bin/env python3
import rospy 
from sensor_msgs.msg import Image 
from cv_bridge import CvBridge, CvBridgeError 
import cv2 
bridge_object = CvBridge() # create the cv_bridge object 
image_received = 0 #Flag to indicate that we have already received an image 
cv_image = 0 #This is just to create the global variable cv_image 

def show_image(): 
    image_sub = rospy.Subscriber("/wamv/sensors/cameras/front_left_camera/image_raw",Image,camera_callback) 
    r = rospy.Rate(10) #10Hz  
    #cv2.waitKey(1) 
    while not rospy.is_shutdown():  
        if image_received: 
            cv2.waitKey(1) 
            r.sleep()  
    cv2.destroyAllWindows() 
    
def camera_callback(data): 
    global bridge_object 
    global cv_image 
    global image_received 
    image_received=1 
    try: 
        print("received ROS image, I will convert it to opencv") 
        # We select bgr8 because its the OpenCV encoding by default 
        cv_image = bridge_object.imgmsg_to_cv2(data, desired_encoding="bgr8") 
        #Add your code to save the image here: 
        #Save the image "img" in the current path  
        #cv2.imwrite('robot_image.jpg', cv_image)        
        #Display the image in a window 
        cv2.imshow('Image from robot camera', cv_image) 
    except CvBridgeError as e: 
        print(e) 
        
if __name__ == '__main__': 
    rospy.init_node('cam_test', anonymous=True) 
    show_image()

`

caguero, 2022-10-28:
Sorry but this is more an OpenCV question than a VRX issue.

closed 2022-10-28 by caguero

YeiSimon, 2023-03-30:
@mazik1496  This is my code, I don't know whether you succeed or not.

`import rospy
import cv2
from sensor_msgs.msg import Image
from std_msgs.msg import Header
from cv_bridge import CvBridge, CvBridgeError
import numpy as np

def capture_video(data):
    
    bridge = CvBridge()
    rospy.loginfo("receiving video frame")
   
    current_frame = bridge.imgmsg_to_cv2(data)
    current_frame = cv2.cvtColor(current_frame, cv2.COLOR_RGB2BGR)
    cv2.imshow("camera", current_frame)
    cv2.waitKey(1)
    
def receive_message():
    rospy.init_node('video_publisher', anonymous=True)
    sub = rospy.Subscriber('/wamv/sensors/cameras/middle_right_camera/image_raw', Image, queue_size=10, callback = capture_video)
    rospy.spin()
 
  # Close down the video stream when done
    cv2.destroyAllWindows()

if __name__ == '__main__':
    try:
        receive_message()
    except rospy.ROSInterruptException:
        pass

`