osrf/vrx #536: How is the ground raised in the map？

cyijie, 2022-11-01:
In vrx_gazebo, I want to raise the land part or lower the water level. To better enable the USV's liDAR to scan the river banks. How should I set up the map?

Thanks for your help！！！！

M1chaelM, 2022-12-01:
This is possible, but it may not give you the effect you want because, unlike real land, our "land" is designed to end at the water. So, raising it up or lowering the water level will leave a gap between land and water in most places. I'm not sure whether this works for your use case, but it's pretty straightforward to test for yourself using the GUI. Here's the steps (assuming you're running the gazebosim branch with ROS2):
1. Start an environment (e.g. Sydney Regatta):
  ```
  ros2 launch vrx_gz competition.launch.py world:=sydney_regatta
  ```
2. Pause the simulation.
3. Expand the "View Angle" menu on the right side.
4. Select one of the cubes that has one of the sides of the cube hightlighted in orange to move the camera to a view level with the water.
5. Click the Select Mode Arrow in the upper left menu.
6. Click on the Sydney Regatta land somewhere to select it. (You should see a white bounding box appear around the whole model.)
7. Click on the Translate Mode button (to the immediate right of the Select Mode Arrow). You should see large green, red and blue arrows appear in the center of the map.
8. Grab the blue arrow and use it to drag the map to the height you want.
9. Unpause the simulation and inspect to check whether this solves your problem.

M1chaelM, 2022-12-09:
If there is no further response in the next few days I will assume this issue has been resolved and close it.

In the meantime, I'll point out that you can also change the position of the map by simply editing the pose tag in the `sdf`. For example, you can edit `words/sydney_regatta.sdf` and look for the element that includes the Sydney Regatta model:
```
<include>
  <!-- Note - the pose tag doesn't seem to work for heightmaps, so you need
      to go into the model file to change the altitude/height! -->
  <pose> 0 0 0.2 0 0 0 </pose>
  <uri>https://fuel.gazebosim.org/1.0/openrobotics/models/sydney_regatta</uri>
</include>
```
Ignore the comment! (It needs to be fixed). You can change the height of the map by changing the z value in the pose to the height that you want. Then rebuild the environment and launch the world:
```
ros2 launch vrx_gz competition.launch.py world:=sydney_regatta
```
Here's a screenshot of the result of running the above command after setting the z value to 1m with `<pose> 0 0 1 0 0 0 </pose>`:

[image]

As you can see, anything more than a very small number will create a gap between the land and the water.

M1chaelM, 2022-12-28:
Closing due to inactivity. Please re-open if your issue is not resolved.

closed 2022-12-28 by M1chaelM