osrf/vrx #538: How do i turn off the wave?

mazik1496, 2022-11-10:
https://github.com/osrf/vrx/blob/master/wave_gazebo/world_models/ocean_waves/model.xacro

I'm in the process of controlling the movement of a ship through a basic example of vrx through roslaunch vrx_gazebo vrx.launch.

All parameters in lines 28 to 30 of the link above have been modified to 0. Nevertheless, there was no change in wave motion.
How can I get rid of wave motion?

bsb808, 2022-11-29:
Have you tried the two options in this tutorial for the "Flat ocean example"? - https://github.com/osrf/vrx/wiki/changing_wave_params_tutorial 

I just verified that both methods work on my setup.

Make sure to run `catkin_make` after making any edits to xacro files.

Also, for option 1 in the tutorial above, change only the `gain` parameter to `0.0` and leave the other values as is.   I'm not sure the behavior with `period:=0`.

caguero, 2023-01-04:
Closing due to inactivity. Feel free to reopen if you provide further information.

closed 2023-01-04 by caguero