osrf/vrx #539: [Garden+Humble] How do you turn the wam-v propeller?

KonradUr, 2022-11-10:
Hi, I am trying to get control of a vessel. Installing the system was OK. I can start it using:
`ros2 launch vrx_gz competition.launch.py world:=sydney_regatta`

It generates the world correctly.
Now I’m trying to move the right thruster:
`ros2 topic pub --once /wamv/thrusters/right/pos std_msgs/msg/Float64 "data: 1.0"`

It works, but when I try to switch on the propeller spinning it fails:
`ros2 topic pub --once /wamv/thrusters/right/thrust std_msgs/msg/Float64 "data: 1.0"`
Nothing happens.
What am I doing wrong?

kizilkayabayram, 2022-11-23:
Did you to set locked to false inside the file ```/vrx_gz/src/vrx_gz/model.py``` ? 
Should be something like this 

```python
...
xacro_command.append(f'locked:=false')
...
```

caguero, 2022-11-29:
Could you verify the version of `libgz-sim7` installed in your system? You should have `7.1.0` or later. If you compiled from source, you should have a version from [this commit](https://github.com/gazebosim/gz-sim/commit/a738dec47ae4f5c18f48a6d4d4b0edb500a490fa) at least (merged November 2nd).

caguero, 2023-01-04:
Closing due to inactivity. Feel free to reopen if you have further information.

closed 2023-01-04 by caguero