osrf/vrx #545: WAMV is not locking before OnRunning

M1chaelM, 2022-11-23:
For many of the tasks ported to gazebosim, the WAMV does not lock properly and can move during the OnReady state. This is definitely the case for the navigation and gymkhana tasks. I will check whether the lock is working for any tasks and update this issue.

caguero, 2022-12-01:
I've just tested the latest `gazebosim` branch and the WAM-V loks/unlocks as expected in all the task worlds. The only world that wasn't working as expected was `sydney_regatta`, as the USV never unlocked. #549 should fix this issue.

@M1chaelM , could you double-check that you get the same behavior? If not, could you remove your `build`, `install` and `log` directories and recompile to see if that makes any difference?

M1chaelM, 2022-12-09:
Confirmed this is working as expected for all tasks. The WAM-V unlocks for OnReady. It's up to the user to avoid driving through the navigation gate before the OnRunning state. To make things a little easier, I will adjust the default course for Navigation and Gymkhana tasks so it is a little further away from the WAM-V starting position.

M1chaelM, 2022-12-10:
Closing because this was not a bug and #554 makes it less likely that a new user will accidentally cross a gate before the plugin is in the "running" state.

closed 2022-12-10 by M1chaelM