osrf/vrx #548: The custom usv doesn't seem to have buoyancy applied

mazik1496, 2022-11-28:
[image]

I want to operate the USV like the picture above in the vrx environment. So, I created urdf.xacro, gazebo.xacro and launch files for this, but it is loaded into the environment in vrx, but buoyancy is not applied, so the phenomenon of continuing to sink occurs. Even if the usv_gazebo_plungin.xacro code parameter is changed, the sinking phenomenon continues to occur. How do I apply buoyancy to my custom usv?? I can send you my code and files as needed.

caguero, 2022-11-29:
Hi @mazik1496 . Probably what's happening is that your model doesn't have the buoyancy plugin attached. You can do that by calling the xacro macro `usv_dynamics_gazebo` declared [here](https://github.com/osrf/vrx/blob/master/wamv_gazebo/urdf/dynamics/wamv_gazebo_dynamics_plugin.xacro).

And [here's](https://github.com/osrf/vrx/blob/master/wamv_gazebo/urdf/dynamics/wamv_gazebo_dynamics_plugin.xacro) the example where we instantiate a copy of this buoyancy plugin attached to the WAM-V.

mazik1496, 2022-11-30:
@caguero I used usv_dynamics_gazebo. but my model doesn't have the buoyancy plugin attached.

caguero, 2022-11-30:
I didn't see it in the xacro code you shared in https://github.com/osrf/vrx/issues/544

mazik1496, 2022-12-01:
@caguero oh sorry After posting that #544, I was looking for a solution by myself, and I mentioned usv_gazebo_plugin in the code after that. My code is as follows. I set all the parameter values to 0 and made them very large, but I found the same gradual decrease. When I saw it, it seems that the link was not properly applied despite the declaration of the link. If you don't mind, I'll give you my code files, so can you check it out??

`<?xml version="1.0"?>
<robot xmlns:xacro="http://ros.org/wiki/xacro">
  <xacro:macro name="usv_dynamics_gazebo" params="name width:=0.9 length:=1.5">
    
    <gazebo>
      <plugin name="usv_dynamics_${name}" filename="libusv_gazebo_dynamics_plugin.so">
        <bodyName>base_link</bodyName>
        
        <waterLevel>0</waterLevel>
        <waterDensity>1000</waterDensity>
        
        <xDotU>0.0</xDotU>
        <yDotV>0.0</yDotV>
        <nDotR>0.0</nDotR>
        
        <xU>120.0</xU>
        <xUU>240.0</xUU>
        <yV>240.0</yV>
        <yVV>2600</yVV>
        <zW>50000.0</zW>
        <zWW>0.0</zWW>
        <kP>50000.0</kP>
        <kPP>5000.0</kPP>
        <mQ>500000.0</mQ>
        <mQQ>5000.0</mQQ>
        <nR>400000.0</nR>
        <nRR>0.0</nRR>
        
        <boatArea>1.35</boatArea>
        <hullRadius>0.5</hullRadius>
        <boatWidth>${width}</boatWidth>
        <boatLength>${length}</boatLength>
        
        <length_n>1</length_n>
        
        <wave_model>ocean_waves</wave_model>
      </plugin>
    </gazebo>
  </xacro:macro>
</robot>`

caguero, 2022-12-01:
Do you see any errors when you launch the simulation?

For example, this error indicates that your link hasn't been found:
```
ROS_INFO_STREAM("Did not find SDF parameter bodyName");
```

The original macro uses:
```
<bodyName>${namespace}/base_link</bodyName>
```
and your macro uses:
```
<bodyName>base_link</bodyName>
```

I also suspect your link is not correctly specified and the plugin can't find it.

mazik1496, 2022-12-02:
So, the part about the namespace was mentioned in the execution launch file and the urdf file, respectively.

launch file:
`<arg name="namespace" default="wamv"/>`

urdf file:
`<xacro:arg name="namespace" default="wamv"/>`

After that, the plug-in also used ${namespace}/base_link as it was, but it was not applied. And while running the launch file, the above error could not be confirmed. That's why I was asking if you could look at my work file if you didn't mind.

mazik1496, 2022-12-05:
@caguero Over the past weekend, I checked it once more while comparing it with the existing vrx code to continue solving it. I checked the answers to my question once again and modified a few things, but the result is the same. Also, at first, I thought I wouldn't be able to upload attachments to questions, but now I think it's possible. So I'm trying to upload a compressed file of the custom usv package I'm working on. Please check it out and let me know if there are any problems.

[oneusv_urdf.zip](https://github.com/osrf/vrx/files/10150847/oneusv_urdf.zip)

closed 2023-01-27 by mazik1496