osrf/vrx #559: usv gazebo dynamics plugin parameter

mazik1496, 2022-12-23:
http://wiki.ros.org/usv_gazebo_plugins

We are analyzing the parameters in the plug-in through the link above. Looking at this, I wonder if there are any reference materials or papers on how variables such as xDotU and xDotV work and where they originate. There is no such explanation in the theory of operation here on the wiki.

M1chaelM, 2022-12-29:
Hi, I think you may be looking for our paper, [Toward Maritime Robotic Simulation in Gazebo](https://wiki.nps.edu/display/BB/Publications?preview=/1173263776/1173263778/PID6131719.pdf), which is linked from the [documentation page](https://github.com/osrf/vrx/wiki/documentation) and the [README of this repository](https://github.com/osrf/vrx).

xDotU and xDotV and other parameters mentioned in the plugin documentation are discussed in the section on the Vehicle Model (p. 5). Does this answer your question?

I'm also tagging @bsb808, who worked on the plugin and this section of the paper.

mazik1496, 2023-01-09:
@M1chaelM
I checked the above document and it is a good resource. thank you! But here another question arises. Starting from page 5, there are various parameters and expressions. I would like to know how these parameters are assigned to the plug-in. For example, I am curious about how the mass changes from 1kg to 5kg, how it is substituted into the plug-in, and where variables such as mass must be set to be substituted into the plug-in.