osrf/vrx #573: Inquiry on Buoyancy plugin

PhilipAmadasun, 2023-01-06:
Hi,
I have been trying to apply functionality to my USV project. By chance, I came across this repo and have been searching for how exactly you guys applied buoyancy. I believe we are primarily using the same gazebo model(you can see my repo [here](https://github.com/PhilipAmadasun/CBS)). I have tried to use the Generic gazebo Buoyancy plugin but this did not work.  For my part, I'm not quite sure what component of the model I should apply the buoyancy plugin to. I tried applying it to the propellers but the model came out disjointed in gazebo. 
I looked through the urdf files in your repo and could not seem to find:
``` [I] WHat Buoyancy Plugin was used

     [ii] WHat components it was used on
```
I would be grateful for any sort of help o this matter.

M1chaelM, 2023-01-06:
Hi,

We have two plugins for buoyancy: one that we use for the vehicle and one for arbitrary objects that float. The one for the vehicle is called `Surface`. The code is in `vrx_gz/src`. Here's [the header](https://github.com/osrf/vrx/blob/main/vrx_gz/src/Surface.hh). To see how it's applied you can look at [the `.erb` file for our vehicle](https://github.com/osrf/vrx/blob/main/vrx_gz/models/wam-v/model.sdf.erb).

That's probably what you're looking for, but just in case: the other plugin is called `PolyhedraBuoyancyDrag`. If you want to see examples of how it's attached, take a look at any of the `sdf` files for models that float. For example, you might look at the buoys in the [obstacle course](https://github.com/osrf/vrx/blob/main/vrx_gz/models/obstacle_course/model.sdf).

PhilipAmadasun, 2023-01-07:
I assume it's the `libSurface.so`. THis might be a dumb questions but:

[I] I searched around and could not find where exactly the .so file was created. Or does the .so file get created when the model loads in gazebo?

[ii] What do I need to use the plugin for my model? Are there any dependencies I need to add to my `CMakeList.txt` file? Or I just copy and paste the `Surface.hh` and `Surface.cc` file?

By the way I want to apply the plugin to my `.urdf.xacro` and `.urdf` file if possible.

As an example, this is my `wam-v.urdf.xacro` file:

```
<?xml version="1.0"?>
<robot xmlns:xacro="http://ros.org/wiki/xacro" name="wam-v">
  <xacro:include filename="properties.urdf.xacro"/>

  <link name="base_link">
    <visual>
     <geometry>
       <mesh filename="package://freddy/mesh/WAM-V.dae"/>
     </geometry>
   </visual>
   <collision name="base_link_collision">
     <geometry>
       <mesh filename="package://freddy/mesh/WAM-V.dae"/>
     </geometry>
   </collision>
   <inertial>
     <mass value="247.07"/>
     <inertia ixx="200" ixy="0.03" ixz="3.09" iyy="500" iyz="-0.05" izz="165.39"/>
   </inertial>
  </link>

  <joint name="base_link_joint" type="fixed">
    <origin xyz="0 0 1.84"/>
    <parent link="base_footprint"/>
    <child link="base_link"/>
  </joint>

  <link name="base_footprint"/>

  <link name="front_camera_link"/>
  <joint name="front_camera_joint" type="fixed">
    <origin xyz="1.2 0 0"/>
    <parent link="base_link"/>
    <child link="front_camera_link"/>
  </joint>
  <link name="left_camera_link"/>
  <joint name="left_camera_joint" type="fixed">
    <origin xyz="0 0.5 0" rpy="0 0 ${pi/2}"/>
    <parent link="base_link"/>
    <child link="left_camera_link"/>
  </joint>
  <link name="right_camera_link"/>
  <joint name="right_camera_joint" type="fixed">
    <origin xyz="0 -0.5 0" rpy="0 0 ${-pi/2}"/>
    <parent link="base_link"/>
    <child link="right_camera_link"/>
  </joint>
  <link name="rear_camera_link"/>
  <joint name="rear_camera_joint" type="fixed">
    <origin xyz="-0.5 0 0" rpy="0 0 ${pi}"/>
    <parent link="base_link"/>
    <child link="rear_camera_link"/>
  </joint>

  <link name="scan"/>
  <joint name="scan_joint" type="fixed">
    <origin xyz="1.5 0 -1" rpy="0 0 0"/>
    <parent link="base_link"/>
    <child link="scan"/>
[... 414 lines cut ...]
               That pixel's noise value is added to each of its color
               channels, which at that point lie in the range [0,1].-->
          <mean>0.0</mean>
          <stddev>0.007</stddev>
        </noise>
      <!--/camera-->
      <plugin name="camera_controller" filename="libgazebo_ros_camera.so">
        <alwaysOn>true</alwaysOn>
        <updateRate>0.0</updateRate>
        <cameraName>wam_v/rear</cameraName>
        <imageTopicName>image_raw</imageTopicName>
        <cameraInfoTopicName>camera_info</cameraInfoTopicName>
        <frameName>front_camera_link</frameName>
        <hackBaseline>0.07</hackBaseline>
        <distortionK1>0.0</distortionK1>
        <distortionK2>0.0</distortionK2>
        <distortionK3>0.0</distortionK3>
        <distortionT1>0.0</distortionT1>
        <distortionT2>0.0</distortionT2>
      </plugin>
    </sensor>
  </gazebo>

</robot>
```

closed 2023-02-16 by PhilipAmadasun