osrf/vrx #574: Building vrx package for ROS melodic went wrong

PhilipAmadasun, 2023-01-15:
I think I somehow broke my gazebo in trying to build the `gazebo-classic` package for ROS melodic. This is what `catkin_make` shows on my terminal.
```
Base path: /home/philip/vrx_ws
Source space: /home/philip/vrx_ws/src
Build space: /home/philip/vrx_ws/build
Devel space: /home/philip/vrx_ws/devel
Install space: /home/philip/vrx_ws/install
Creating symlink "/home/philip/vrx_ws/src/CMakeLists.txt" pointing to "/opt/ros/melodic/share/catkin/cmake/toplevel.cmake"
####
#### Running command: "cmake /home/philip/vrx_ws/src -DCATKIN_DEVEL_PREFIX=/home/philip/vrx_ws/devel -DCMAKE_INSTALL_PREFIX=/home/philip/vrx_ws/install -G Unix Makefiles" in "/home/philip/vrx_ws/build"
####
-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using CATKIN_DEVEL_PREFIX: /home/philip/vrx_ws/devel
-- Using CMAKE_PREFIX_PATH: /home/philip/coverage_ws/devel;/opt/ros/melodic
-- This workspace overlays: /home/philip/coverage_ws/devel;/opt/ros/melodic
-- Found PythonInterp: /usr/bin/python2 (found suitable version "2.7.17", minimum required is "2") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python2
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/philip/vrx_ws/build/test_results
-- Found gtest sources under '/usr/src/googletest': gtests will be built
-- Found gmock sources under '/usr/src/googletest': gmock will be built
CMake Deprecation Warning at /usr/src/googletest/CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

CMake Deprecation Warning at /usr/src/googletest/googlemock/CMakeLists.txt:41 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

CMake Deprecation Warning at /usr/src/googletest/googletest/CMakeLists.txt:48 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Found PythonInterp: /usr/bin/python2 (found version "2.7.17") 
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.29
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
[... 695 lines cut ...]
gzclient: symbol lookup error: /usr/lib/x86_64-linux-gnu/libgazebo_common.so.9: undefined symbol: _ZN8ignition10fuel_tools12ClientConfig12SetUserAgentERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE
gzserver: symbol lookup error: /usr/lib/x86_64-linux-gnu/libgazebo_common.so.9: undefined symbol: _ZN8ignition10fuel_tools12ClientConfig12SetUserAgentERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE
[gazebo-2] process has died [pid 23561, exit code 127, cmd /opt/ros/melodic/lib/gazebo_ros/gzserver -e ode /opt/ros/melodic/share/turtlebot3_gazebo/worlds/empty.world __name:=gazebo __log:=/home/philip/.ros/log/7b851758-9501-11ed-833a-d43b04207196/gazebo-2.log].
log file: /home/philip/.ros/log/7b851758-9501-11ed-833a-d43b04207196/gazebo-2*.log
[gazebo_gui-3] process has died [pid 23567, exit code 127, cmd /opt/ros/melodic/lib/gazebo_ros/gzclient __name:=gazebo_gui __log:=/home/philip/.ros/log/7b851758-9501-11ed-833a-d43b04207196/gazebo_gui-3.log].
log file: /home/philip/.ros/log/7b851758-9501-11ed-833a-d43b04207196/gazebo_gui-3*.log
^C[spawn_urdf-4] killing on exit
Traceback (most recent call last):
  File "/opt/ros/melodic/lib/gazebo_ros/spawn_model", line 239, in <module>
    exit_code = sm.run()
  File "/opt/ros/melodic/lib/gazebo_ros/spawn_model", line 185, in run
    self.args.gazebo_namespace)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/gazebo_ros/gazebo_interface.py", line 28, in spawn_urdf_model_client
    rospy.wait_for_service(gazebo_namespace+'/spawn_urdf_model')
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 166, in wait_for_service
    raise ROSInterruptException("rospy shutdown")
rospy.exceptions.ROSInterruptException: rospy shutdown
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shuttin
```

Is there a way to fix this without completely removing ROS and gazebo from my system and reinstalling everything?
I use ubuntu 18.04

PhilipAmadasun, 2023-01-16:
Okay so after upgrading certain libraries my gazebo seems to work again. But on trying to `catkin_make` I now get:
```
In file included from /usr/include/gazebo-9/gazebo/common/common.hh:8:0,
                 from /usr/include/gazebo-9/gazebo/gazebo_core.hh:19,
                 from /usr/include/gazebo-9/gazebo/gazebo.hh:20,
                 from /home/philip/vrx_ws/src/vrx/wave_gazebo_plugins/src/Utilities.cc:22:
/usr/include/gazebo-9/gazebo/common/Battery.hh:25:10: fatal error: sdf/sdf.hh: No such file or directory
 #include "sdf/sdf.hh"
          ^~~~~~~~~~~~
In file included from /usr/include/gazebo-9/gazebo/common/common.hh:8:0,
                 from /usr/include/gazebo-9/gazebo/gazebo_core.hh:19,
                 from /usr/include/gazebo-9/gazebo/gazebo.hh:20,
                 from /home/philip/vrx_ws/src/vrx/wave_gazebo_plugins/src/Gazebo.cc:23:
/usr/include/gazebo-9/gazebo/common/Battery.hh:25:10: fatal error: sdf/sdf.hh: No such file or directory
 #include "sdf/sdf.hh"
          ^~~~~~~~~~~~
compilation terminated.
compilation terminated.
vrx/wave_gazebo_plugins/CMakeFiles/Hydrodynamics.dir/build.make:75: recipe for target 'vrx/wave_gazebo_plugins/CMakeFiles/Hydrodynamics.dir/src/Gazebo.cc.o' failed
make[2]: *** [vrx/wave_gazebo_plugins/CMakeFiles/Hydrodynamics.dir/src/Gazebo.cc.o] Error 1
vrx/wave_gazebo_plugins/CMakeFiles/Hydrodynamics.dir/build.make:131: recipe for target 'vrx/wave_gazebo_plugins/CMakeFiles/Hydrodynamics.dir/src/Utilities.cc.o' failed
make[2]: *** [vrx/wave_gazebo_plugins/CMakeFiles/Hydrodynamics.dir/src/Utilities.cc.o] Error 1
In file included from /usr/include/gazebo-9/gazebo/physics/Link.hh:25:0,
                 from /home/philip/vrx_ws/src/vrx/vrx_gazebo/src/ball_shooter_plugin.cc:19:
/usr/include/gazebo-9/gazebo/msgs/msgs.hh:22:10: fatal error: sdf/sdf.hh: No such file or directory
 #include <sdf/sdf.hh>
          ^~~~~~~~~~~~
compilation terminated.
vrx/vrx_gazebo/CMakeFiles/ball_shooter_plugin.dir/build.make:89: recipe for target 'vrx/vrx_gazebo/CMakeFiles/ball_shooter_plugin.dir/src/ball_shooter_plugin.cc.o' failed
make[2]: *** [vrx/vrx_gazebo/CMakeFiles/ball_shooter_plugin.dir/src/ball_shooter_plugin.cc.o] Error 1
CMakeFiles/Makefile2:5076: recipe for target 'vrx/vrx_gazebo/CMakeFiles/ball_shooter_plugin.dir/all' failed
make[1]: *** [vrx/vrx_gazebo/CMakeFiles/ball_shooter_plugin.dir/all] Error 2
In file included from /usr/include/gazebo-9/gazebo/common/common.hh:8:0,
                 from /usr/include/gazebo-9/gazebo/gazebo_core.hh:19,
                 from /usr/include/gazebo-9/gazebo/gazebo.hh:20,
                 from /home/philip/vrx_ws/src/vrx/wave_gazebo_plugins/src/Wavefield.cc:25:
/usr/include/gazebo-9/gazebo/common/Battery.hh:25:10: fatal error: sdf/sdf.hh: No such file or directory
 #include "sdf/sdf.hh"
          ^~~~~~~~~~~~
compilation terminated.
vrx/wave_gazebo_plugins/CMakeFiles/Hydrodynamics.dir/build.make:145: recipe for target 'vrx/wave_gazebo_plugins/CMakeFiles/Hydrodynamics.dir/src/Wavefield.cc.o' failed
make[2]: *** [vrx/wave_gazebo_plugins/CMakeFiles/Hydrodynamics.dir/src/Wavefield.cc.o] Error 1
In file included from /usr/include/gazebo-9/gazebo/physics/ModelState.hh:28:0,
                 from /usr/include/gazebo-9/gazebo/physics/Model.hh:29,
                 from /usr/include/gazebo-9/gazebo/physics/Actor.hh:26,
                 from /usr/include/gazebo-9/gazebo/physics/physics.hh:2,
                 from /home/philip/vrx_ws/src/vrx/wave_gazebo_plugins/src/WavefieldEntity.cc:22:
/usr/include/gazebo-9/gazebo/physics/State.hh:26:10: fatal error: sdf/sdf.hh: No such file or directory
 #include <sdf/sdf.hh>
          ^~~~~~~~~~~~
compilation terminated.
vrx/wave_gazebo_plugins/CMakeFiles/Hydrodynamics.dir/build.make:159: recipe for target 'vrx/wave_gazebo_plugins/CMakeFiles/Hydrodynamics.dir/src/WavefieldEntity.cc.o' failed
make[2]: *** [vrx/wave_gazebo_plugins/CMakeFiles/Hydrodynamics.dir/src/WavefieldEntity.cc.o] Error 1
CMakeFiles/Makefile2:2883: recipe for target 'vrx/wave_gazebo_plugins/CMakeFiles/Hydrodynamics.dir/all' failed
make[1]: *** [vrx/wave_gazebo_plugins/CMakeFiles/Hydrodynamics.dir/all] Error 2
Makefile:145: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j12 -l12" failed
```

azunyan-nyan-nyan, 2023-01-31:
I've had the same error while using different versions of ros.  I recommend using noetic with ubuntu 20.04 focal, since the ros commands given in the tutorials regarding gazebo_classic are all noetic, and that noetic is said to work best with focal.

caguero, 2023-01-31:
Indeed,  the official configuration for VRX with Gazebo classic is Ubuntu Focal and ROS Noetic. See https://github.com/osrf/vrx/wiki/system_requirements .

closed 2023-01-31 by caguero