osrf/vrx #577: fatal: Remote branch gazebosim not found in upstream origin

azunyan-nyan-nyan, 2023-01-21:
During setup in 22.04.1 Jammy, after installing ROS 2 Humble and Gazebo Garden

`mkdir -p ~/vrx_ws/src`
`cd ~/vrx_ws/src`
`git clone https://github.com/osrf/vrx.git -b gazebosim`

fails to execute because gazebosim isn't found when cloning into vrx

pantelis, 2023-01-23:
We also face the same issue. Are there any updates or work arounds ?

azunyan-nyan-nyan, 2023-01-31:
> We also face the same issue. Are there any updates or work arounds ?

i didn't have any issues when i just used Noetic with Gazebo Classic on Ubuntu 20.04 (Focal Fossa)

azunyan-nyan-nyan, 2023-01-31:
has to be an issue with branching, there is no gazeboism branch listed, although the next command for humble does.

the only available gazebo branch on their github is shown in the below query:

https://github.com/osrf/vrx/branches/all?query=gazebo

and it's gazebo_classic, which still works for simulation (from what i've seen so far)

atticusrussell, 2023-01-31:
I just encountered the same issue following these directions:
https://github.com/osrf/vrx/wiki/2023-installation
anyone have correct instructions for 22.04 and Humble ROS?

caguero, 2023-01-31:
Sorry, that was an issue in the tutorial. We forgot to update it when we moved the branch from `gazebosim` to `main` as you already mentioned. It's fixed in the tutorial already. Thanks!

closed 2023-01-31 by caguero