osrf/vrx #580: "gflags" missing during colcon build

atticusrussell, 2023-01-31:
following the wiki https://github.com/osrf/vrx/wiki/2023-installation
on a fresh install I get an error from the following step: GZ_VERSION=garden colcon build --merge-install

I am encountering the following error

```
rosdev@devcraft:~/vrx_ws$` GZ_VERSION=garden colcon build --merge-install
Starting >>> ros_gz_interfaces
Starting >>> ros_gz_sim
Starting >>> vrx_gazebo
Finished <<< vrx_gazebo [0.44s]
Starting >>> wamv_description
Finished <<< wamv_description [0.12s]
Starting >>> wamv_gazebo
Finished <<< wamv_gazebo [0.14s]
Finished <<< ros_gz_interfaces [1.07s]
Starting >>> ros_gz_bridge
Starting >>> ros_ign_interfaces
--- stderr: ros_gz_sim
/home/dev15440/vrx_ws/src/ros_gz/ros_gz_sim/src/create.cpp:15:10: fatal error: gflags/gflags.h: No such file or directory
   15 | #include <gflags/gflags.h>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [CMakeFiles/create.dir/build.make:76: CMakeFiles/create.dir/src/create.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:159: CMakeFiles/create.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< ros_gz_sim [7.97s, exited with code 2]
Aborted  <<< ros_ign_interfaces [35.2s]
Aborted  <<< ros_gz_bridge [37.5s]

Summary: 4 packages finished [38.8s]
  1 package failed: ros_gz_sim
  2 packages aborted: ros_gz_bridge ros_ign_interfaces
  2 packages had stderr output: ros_gz_bridge ros_gz_sim
  10 packages not processed
```

atticusrussell, 2023-01-31:
I fixed my issue by running:
$```sudo apt install libgflags-dev```
to get the missing library

now it colcon successfully builds with several stderr output notes about ignition namespace being deprecated, and only the warning:
```
Finished <<< ros_gz [0.96s]
Finished <<< ros_ign_gazebo_demos [1.01s]
Starting >>> ros_ign
Finished <<< ros_ign [1.25s]
--- stderr: vrx_gz
CMake Warning (dev) at CMakeLists.txt:85 (add_library):
  Policy CMP0115 is not set: Source file extensions must be explicit.  Run
  "cmake --help-policy CMP0115" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.

  File:

    /home/dev15440/vrx_ws/src/vrx/vrx_gz/src/WaypointMarkers.cc
This warning is for project developers.  Use -Wno-dev to suppress it.

---
Finished <<< vrx_gz [26.3s]

Summary: 17 packages finished [3min 43s]
  4 packages had stderr output: ros_gz_bridge ros_gz_image ros_gz_sim vrx_gz
```

YeiSimon, 2023-03-11:
while following the installation, I get an error with  GZ_VERSION=garden colcon build --merge-install
[image]
with step I missing?

tiko5000, 2023-03-14:
@YeiSimon: try: `sudo apt-get install ros-humble-xacro`

caguero, 2025-05-16:
We updated the code and I think this shouldn't be an issue. Please, reopen if necessary.

closed 2025-05-16 by caguero