osrf/vrx #586: twist2thrust.py works, but it doesn't seem to be 100% functional.

mazik1496, 2023-02-09:
I'm in the process of building my custom USV. Currently, the custom USV is launched in the VRX virtual environment, and the inertia of the STL files I use is also applied. Since then, I am using twist2thrust.py to control the USV. Currently, the function is operating, but if I hold down a key originally, the USV should continue to move in that direction, but in the case of my USV, it stops while running. I would like to know what could be the cause of this. The custom USV code file I worked on is attached below. (Please understand that I have uploaded attachments because there are too many to attach the entire code contents)

[onekw_description.zip](https://github.com/osrf/vrx/files/10694959/onekw_description.zip)

mazik1496, 2023-02-09:
It seems to be a problem with the current code, so we are working on fixing it. We will close this issue once the issue is fully resolved.

caguero, 2025-05-08:
twist2thrust no longer used. Please, reopen if necessary.

closed 2025-05-08 by caguero