osrf/vrx #589: python3-sdformat13 installation

pantelis, 2023-02-10:
We are attempting to install VRX 2023 version in a `FROM osrf/ros:humble-desktop-full` docker container and we fail to understand how to install the last instruction in the [wiki](http://sdformat.org/tutorials?tut=install&ver=%3E%3D+12&cat=developers&) that concerns python3-sdformat13. The package is not available in Ubuntu 22.04. 

Do we need to follow alternative binary installation instructions as is shown in http://sdformat.org/tutorials?tut=install&ver=%3E%3D+12&cat=developers&#ubuntu ?

```
sudo apt-get install python3-pip wget lsb-release gnupg curl
sudo sh -c 'echo "deb http://packages.osrfoundation.org/gazebo/ubuntu-stable `lsb_release -cs` main" > /etc/apt/sources.list.d/gazebo-stable.list'
wget http://packages.osrfoundation.org/gazebo.key -O - | sudo apt-key add -
sudo apt-get update

sudo apt install libsdformat<#>-dev libsdformat<#>
```

If these are the right instructions, will those install the python3-sdformat13 also ? 

Thank you

M1chaelM, 2023-03-08:
I'm able to install the package locally on Ubuntu 22 by running `sudo apt install python3-sdformat13`, so I think the package should be available in Ubuntu 22 and you should not have to follow any alternative instructions to install it.

Do you have a Dockerfile you could post?

M1chaelM, 2023-03-14:
I created a Dockerfile to test this using ros:humble-desktop-base-jammy. It seems to work, but you do have to run `apt update` before you install the package:
```
$ sudo apt install python3-sdformat13
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package python3-sdformat13
$ sudo apt update
...
$ sudo apt install python3-sdformat13
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following NEW packages will be installed:
  python3-sdformat13
0 upgraded, 1 newly installed, 0 to remove and 28 not upgraded.
Need to get 323 kB of archives.
After this operation, 1615 kB of additional disk space will be used.
Get:1 http://packages.osrfoundation.org/gazebo/ubuntu-stable jammy/main amd64 python3-sdformat13 amd64 13.4.1-1~jammy [323 kB]
Fetched 323 kB in 0s (722 kB/s)            
debconf: delaying package configuration, since apt-utils is not installed
Selecting previously unselected package python3-sdformat13.
(Reading database ... 143845 files and directories currently installed.)
Preparing to unpack .../python3-sdformat13_13.4.1-1~jammy_amd64.deb ...
Unpacking python3-sdformat13 (13.4.1-1~jammy) ...
Setting up python3-sdformat13 (13.4.1-1~jammy) ...
```

M1chaelM, 2023-03-23:
Closing due to inactivity. Please reopen if the issue has not been resolved.

closed 2023-03-23 by M1chaelM