osrf/vrx #595: How to create USV Model for buoyancy plugin.

PhilipAmadasun, 2023-02-16:
In what way would I have to create my custom USV model such that the buoyancy plugin created in this package can be used?

M1chaelM, 2023-02-24:
The buoyancy model we use for our default USV is implemented via our [Surface plugin](https://github.com/osrf/vrx/blob/main/vrx_gz/src/Surface.hh), which simulates buoyancy of objects at the surface of a fluid. Since we have a two-hulled vehicle, we [attach an instance to each hull](https://github.com/osrf/vrx/blob/main/vrx_gz/models/wam-v/model.sdf.erb#L159). The plugin assumes a cylindrical hull by default, but you could override this behavior if you need to.

So, to answer your question, the simplest approach is probably to create a vehicle with cylindrical hulls and attach the Surface plugin in the same way we do. However, this is not an absolute requirement.

Does this answer resolve the issue for you?

PhilipAmadasun, 2023-02-27:
@M1chaelM I'm new to plugins, how does `libsurface.so` know to link itself to the `surface.hh` code?

M1chaelM, 2023-02-27:
`libsurface.so` is the binary produced by compiling `surface.hh` and `surface.cc` (which happens during the build process). Once it is created you don't need the source code to run it. I don't think you should have to make any changes to the current repository to get libsurface.so to build correctly.

That said, I'm not completely sure I've understood your question--is there an error message you're trying to resolve?

PhilipAmadasun, 2023-03-03:
@M1chaelM  I think you answered my question well enough, thank you. I have some more questions,. For one, What parts of this package do I specifically need if I just wanted your buoyancy and thruster plugins for my model?

M1chaelM, 2023-03-03:
Ok, sure. Could you give me a high-level explanation of what you're trying to do? This will help me answer your questions.

PhilipAmadasun, 2023-03-11:
@M1chaelM  I'll try to make a comprehensive writeup this weekend. Please don't close this issue thread

PhilipAmadasun, 2023-03-12:
@M1chaelM    
I want to create a USV model in gazebo with functionality to float and move in water.
I will create a model that will be made of the following components.

[i] Chassis: this is hold the sensors
[ii] Hulls: One on each side of the chassis. They will be the floaters
[iii] Propellers: One attached to each chassis

To provide the floating functionality to the USV. I must provide the Hulls and propellers a buoyancy plugin.
To provide the thruster functionality. I must provide  the propollers with a thruster plugin.
I do that in the urdf file as you know. 

You guys created custom thruster and buoyancy plugins. What part of your package do I need if I only want those plugins? I clearly don’t need to build your entire package if I only need these two specific things.

M1chaelM, 2023-03-14:
OK, thanks for these updates. It sounds like you want to build a USV model that works in another simulation that is already simulating the surface of the water. Is that right? If so, what are you using to create the environment that you want the USV to operate in?

PhilipAmadasun, 2023-03-14:
@M1chaelM  I found a script of a world I was going to use. However I haven't thought about "simulating the surface of water", I just have a world with a lake in it. DOes the water itself require it's own plugin?

PhilipAmadasun, 2023-03-29:
@M1chaelM  Does my question make sense?

M1chaelM, 2023-04-03:
@PhilipAmadasun It sounds like you are working on developing another simulation environment for maritime vehicles. Can you explain why you are doing this so I understand what you're trying to achieve? I suspect this is beyond the scope of what we can support in this space, but I'll leave the issue open for your reply.

PhilipAmadasun, 2023-04-04:
@M1chaelM  I don't have any particular reason for developing another world. If there is a way for me to simulate my USV model in your simulated environments  using the plugins that you guys created, I'll just do that. Anything that allows me to use your plugins for my model is the direction I want to go.

PhilipAmadasun, 2023-05-07:
So in 
 ```
<xacro:macro name="usv_dynamics_gazebo" params="name wavefield_size:=1000">
    <!--Gazebo Plugin for simulating WAM-V dynamics-->
    <gazebo>
      <!-- Left hull -->
      <plugin
        filename="libSurface.so"
        name="vrx::Surface">
        <link_name>${namespace}/base_link</link_name>
        <hull_length>4.9</hull_length>
        <hull_radius>0.213</hull_radius>
        <fluid_level>0</fluid_level>
        <points>
          <point>0.6 1.03 0</point>
          <point>-1.4 1.03 0</point>
        </points>
        <wavefield>
          <topic>/vrx/wavefield/parameters</topic>
        </wavefield>
      </plugin>

      <!-- Right hull -->
      <plugin
        filename="libSurface.so"
        name="vrx::Surface">
        <link_name>${namespace}/base_link</link_name>
        <hull_length>4.9</hull_length>
        <hull_radius>0.213</hull_radius>
        <fluid_level>0</fluid_level>
        <points>
          <point>0.6 -1.03 0</point>
          <point>-1.4 -1.03 0</point>
        </points>
        <wavefield>
          <topic>/vrx/wavefield/parameters</topic>
        </wavefield>
      </plugin>

      <plugin
        filename="libSimpleHydrodynamics.so"
        name="vrx::SimpleHydrodynamics">
        <link_name>${namespace}/base_link</link_name>
        <!-- Added mass -->
        <xDotU>0.0</xDotU>
        <yDotV>0.0</yDotV>
        <nDotR>0.0</nDotR>
        <!-- Linear and quadratic drag -->
        <xU>51.3</xU>
        <xUU>72.4</xUU>
        <yV>40.0</yV>
        <yVV>0.0</yVV>
        <zW>500.0</zW>
        <kP>100.0</kP>
        <mQ>100.0</mQ>
        <nR>400.0</nR>
        <nRR>0.0</nRR>
      </plugin>

    </gazebo>
  </xacro:macro>
```
I dont know understand how the <linkname> you use is the base link yet you say you attch the instance to a hall. Also what is the <point> tag?

PhilipAmadasun, 2023-05-13:
@M1chaelM  I think I misundertood how you guys set up your USV model. The chassis and pontoons(floaters) are one model correct? If so tht's what the "hall" is right? I still have a hard figuring out just how to implement your plugins in my urdf.xacro file so I will copy and paste my urdf.xacro file here with a picture of my model in rviz. 

**My model**
**Description**
My model is made of  3 components:
[i] The chassis and two floaters are one model. This model is called "chassis".
[ii] There is the right and left motor connected to the base via "fixed" joints respectively.
[iii] The right and left propellers are connected to the each motor via "revolute" joints

[image]

[image]

PhilipAmadasun, 2023-05-13:
This is my urdf.xacro file. How would you implement your bouyancy and thruster plugins to this file:
```
<?xml version="1.0" encoding="utf-8"?>
<!-- =================================================================================== -->
<!-- |    This document was autogenerated by xacro from microfred10.urdf               | -->
<!-- |    EDITING THIS FILE BY HAND IS NOT RECOMMENDED                                 | -->
<!-- =================================================================================== -->
<!-- This URDF was automatically created by SolidWorks to URDF Exporter! Originally created by Stephen Brawner (brawner@gmail.com) 
     Commit Version: 1.6.0-4-g7f85cfe  Build Version: 1.6.7995.38578
     For more information, please see http://wiki.ros.org/sw_urdf_exporter -->
<robot name="microfred10">
  <link name="chassis">
    <inertial>
      <origin rpy="0 0 0" xyz="0.00025032 0.0064565 0.036633"/>
      <mass value="5.2783"/>
      <inertia ixx="0.073106" ixy="1.4003E-05" ixz="-2.6555E-08" iyy="0.0038771" iyz="-0.00072596" izz="0.075169"/>
    </inertial>
    <visual>
      <origin rpy="0 0 0" xyz="0 0 0"/>
      <geometry>
        <mesh filename="package://microfred10/meshes/chassis.STL"/>
      </geometry>
      <material name="">
        <color rgba="0.75294 0.75294 0.75294 1"/>
      </material>
    </visual>
    <collision>
      <origin rpy="0 0 0" xyz="0 0 0"/>
      <geometry>
        <mesh filename="package://microfred10/meshes/chassis.STL"/>
      </geometry>
    </collision>
  </link>
  <link name="left_motor">
    <inertial>
      <origin rpy="0 0 0" xyz="1.3878E-17 3.6028E-06 -4.0126E-05"/>
      <mass value="0.0064969"/>
      <inertia ixx="4.6382E-07" ixy="-3.8938E-23" ixz="-8.3406E-22" iyy="5.1206E-07" iyz="-2.9618E-10" izz="2.6945E-07"/>
    </inertial>
    <visual>
      <origin rpy="0 0 0" xyz="0 0 0"/>
      <geometry>
        <mesh filename="package://microfred10/meshes/left_motor.STL"/>
      </geometry>
      <material name="">
        <color rgba="0.79216 0.81961 0.93333 1"/>
      </material>
    </visual>
    <collision>
      <origin rpy="0 0 0" xyz="0 0 0"/>
      <geometry>
        <mesh filename="package://microfred10/meshes/left_motor.STL"/>
      </geometry>
    </collision>
  </link>
  <joint name="left_motor_joint" type="fixed">
    <origin rpy="1.2217 0 3.1416" xyz="-0.098718 -0.079178 0.069748"/>
    <parent link="chassis"/>
    <child link="left_motor"/>
    <axis xyz="0 0 0"/>
  </joint>
  <link name="left_prop">
    <inertial>
      <origin rpy="0 0 0" xyz="-2.6043E-09 1.4741E-09 -0.053213"/>
      <mass value="0.0022027"/>
      <inertia ixx="1.9639E-06" ixy="-5.1712E-15" ixz="-3.885E-13" iyy="1.9639E-06" iyz="2.4387E-13" izz="3.1575E-08"/>
    </inertial>
    <visual>
      <origin rpy="0 0 0" xyz="0 0 0"/>
      <geometry>
        <mesh filename="package://microfred10/meshes/left_prop.STL"/>
      </geometry>
      <material name="">
        <color rgba="0.75294 0.75294 0.75294 1"/>
      </material>
    </visual>
    <collision>
      <origin rpy="0 0 0" xyz="0 0 0"/>
      <geometry>
        <mesh filename="package://microfred10/meshes/left_prop.STL"/>
      </geometry>
    </collision>
  </link>
  <joint name="left_prop_joint" type="revolute">
    <origin rpy="0 0 1.5952" xyz="0 0 -0.0051837"/>
    <parent link="left_motor"/>
    <child link="left_prop"/>
    <axis xyz="0 0 1"/>
    <limit effort="1" lower="0" upper="365" velocity="840"/>
  </joint>
  <link name="right_motor">
    <inertial>
      <origin rpy="0 0 0" xyz="0 3.6028E-06 -4.0126E-05"/>
      <mass value="0.0064969"/>
      <inertia ixx="4.6382E-07" ixy="-3.6347E-24" ixz="-4.3461E-22" iyy="5.1206E-07" iyz="-2.9618E-10" izz="2.6945E-07"/>
    </inertial>
    <visual>
      <origin rpy="0 0 0" xyz="0 0 0"/>
      <geometry>
        <mesh filename="package://microfred10/meshes/right_motor.STL"/>
      </geometry>
      <material name="">
        <color rgba="0.79216 0.81961 0.93333 1"/>
      </material>
    </visual>
    <collision>
      <origin rpy="0 0 0" xyz="0 0 0"/>
      <geometry>
        <mesh filename="package://microfred10/meshes/right_motor.STL"/>
      </geometry>
    </collision>
  </link>
  <joint name="right_motor_joint" type="fixed">
    <origin rpy="1.2217 0 3.1416" xyz="0.098718 -0.079178 0.069748"/>
    <parent link="chassis"/>
    <child link="right_motor"/>
    <axis xyz="0 0 0"/>
  </joint>
  <link name="right_prop">
    <inertial>
      <origin rpy="0 0 0" xyz="2.4847E-09 2.0796E-09 -0.053213"/>
      <mass value="0.0022027"/>
      <inertia ixx="1.9639E-06" ixy="-4.7676E-15" ixz="3.8518E-13" iyy="1.9639E-06" iyz="3.0399E-13" izz="3.1575E-08"/>
    </inertial>
    <visual>
      <origin rpy="0 0 0" xyz="0 0 0"/>
      <geometry>
        <mesh filename="package://microfred10/meshes/right_prop.STL"/>
      </geometry>
      <material name="">
        <color rgba="0.75294 0.75294 0.75294 1"/>
      </material>
    </visual>
    <collision>
      <origin rpy="0 0 0" xyz="0 0 0"/>
      <geometry>
        <mesh filename="package://microfred10/meshes/right_prop.STL"/>
      </geometry>
    </collision>
  </link>
  <joint name="right_prop_joint" type="revolute">
    <origin rpy="0 0 -1.5952" xyz="0 0 -0.0051837"/>
    <parent link="right_motor"/>
    <child link="right_prop"/>
    <axis xyz="0 0 1"/>
    <limit effort="1" lower="0" upper="365" velocity="840"/>
  </joint>
</robot>
```

PhilipAmadasun, 2023-05-18:
@M1chaelM  Maybe there is better to have the floaters/hulls  be their own model? If so, I have that urdf file as well. 
Here is the urdf.xacro file, I then tried to include the xacro file thatt includes the boyancy plugin.
```

<?xml version="1.0" encoding="utf-8"?>
<!-- =================================================================================== -->
<!-- |    This document was autogenerated by xacro from microfred5_gazebo.urdf         | -->
<!-- |    EDITING THIS FILE BY HAND IS NOT RECOMMENDED                                 | -->
<!-- =================================================================================== -->
<!-- This URDF was automatically created by SolidWorks to URDF Exporter! Originally created by Stephen Brawner (brawner@gmail.com) 
     Commit Version: 1.6.0-4-g7f85cfe  Build Version: 1.6.7995.38578
     For more information, please see http://wiki.ros.org/sw_urdf_exporter prop=volume is 0.00000220, center of mass=(0 0 -0.034647), pontoon=volume 0.00160797  center of mass=(0.0076 2-->
<robot name="microfred5">
  <link name="Chassis">
    <inertial>
      <origin rpy="0 0 0" xyz="-0.000866333072603661 -0.000465181809646093 0.0936581765573819"/>
      <mass value="0.553562943068425"/>
      <inertia ixx="0.000759295112460544" ixy="9.04864829224519E-06" ixz="-1.68708778955156E-06" iyy="0.00166934730287536" iyz="-6.43084689336098E-07" izz="0.00240189000215462"/>
    </inertial>
    <visual>
      <origin rpy="0 0 0" xyz="0 0 0"/>
      <geometry>
        <mesh filename="package://microfred5/meshes/Chassis.STL"/>
      </geometry>
      <material name="">
        <color rgba="0.752941176470588 0.752941176470588 0.752941176470588 1"/>
      </material>
    </visual>
    <collision>
      <origin rpy="0 0 0" xyz="0 0 0"/>
      <geometry>
        <mesh filename="package://microfred5/meshes/Chassis.STL"/>
      </geometry>
    </collision>
  </link>
  <link name="front_camera_link">
    <collision>
      <origin rpy="0 0 0" xyz="0 0 0"/>
      <geometry>
        <box size="0.01 0.01 0.01"/>
      </geometry>
    </collision>
    <visual>
      <origin rpy="0 0 0" xyz="0 0 0"/>
      <geometry>
        <box size="0.01 0.01 0.01"/>
      </geometry>
      <material name="red"/>
    </visual>
    <inertial>
      <mass value="1e-5"/>
      <origin rpy="0 0 0" xyz="0 0 0"/>
      <inertia ixx="1e-6" ixy="0" ixz="0" iyy="1e-6" iyz="0" izz="1e-6"/>
    </inertial>
  </link>
  <joint name="front_camera_joint" type="fixed">
    <origin xyz="0.08 0 0.155"/>
    <parent link="Chassis"/>
    <child link="front_camera_link"/>
    <axis xyz="0 0 0"/>
  </joint>
  <link name="Left_Pontoon">
    <inertial>
      <origin rpy="0 0 0" xyz="-0.0565745716650207 0.00984376682513155 -0.0221303741797096"/>
      <mass value="2.30097036634083"/>
      <inertia ixx="0.00153207872976563" ixy="-1.00799724109655E-05" ixz="-0.00036224874205453" iyy="0.035607611518654" iyz="3.5819734102809E-07" izz="0.0362486131418558"/>
    </inertial>
    <visual>
      <origin rpy="0 0 0" xyz="0 0 0"/>
      <geometry>
[... 159 lines cut ...]
        <xU>51.3</xU>
        <xUU>72.4</xUU>
        <yV>40.0</yV>
        <yVV>0.0</yVV>
        <zW>500.0</zW>
        <kP>100.0</kP>
        <mQ>100.0</mQ>
        <nR>400.0</nR>
        <nRR>0.0</nRR>
      </plugin>

    </gazebo>
</xacro:macro>
```
It doesn't work. I don't understand this part:
```       <plugin
        filename="libSimpleHydrodynamics.so"
        name="vrx::SimpleHydrodynamics"> .....
``` 
I dont understand what link of the robot I'm supposed to place for that part.

This is repo os very sophisticated and so is really confusing, there are so many xacro: macro files used in different directories. 
Is there a way to just directly add everything I need to just one `.urdf.xacro` file? Let me just start with something simple. 

My goal is to place my robot in an empty gazebo world with the Bouyancy and Thruster Plugins workig correctly. I would be grateful for help on this.

PhilipAmadasun, 2023-05-18:
I use ROS melodic by the way incase this is useful information

PhilipAmadasun, 2023-05-19:
@M1chaelM Thanks for the reply. Unfortunately my team is really noew to ROS so we are just using ROS melodic and noetic. Is it at all possible to create a tutorial for those ROS versions?

PhilipAmadasun, 2023-05-28:
@M1chaelM  I tried using the `gazebo_classic` version of the bouyancy plugin. This is the urdf.xacro file no: 
```
<?xml version="1.0" encoding="utf-8"?>
<!-- =================================================================================== -->
<!-- |    This document was autogenerated by xacro from microfred10.urdf               | -->
<!-- |    EDITING THIS FILE BY HAND IS NOT RECOMMENDED                                 | -->
<!-- =================================================================================== -->
<!-- This URDF was automatically created by SolidWorks to URDF Exporter! Originally created by Stephen Brawner (brawner@gmail.com) 
     Commit Version: 1.6.0-4-g7f85cfe  Build Version: 1.6.7995.38578
     For more information, please see http://wiki.ros.org/sw_urdf_exporter -->
<robot name="microfred10">
  <link name="chassis">
    <inertial>
      <origin rpy="0 0 0" xyz="0.00025032 0.0064565 0.036633"/>
      <mass value="5.2783"/>
      <inertia ixx="0.073106" ixy="1.4003E-05" ixz="-2.6555E-08" iyy="0.0038771" iyz="-0.00072596" izz="0.075169"/>
    </inertial>
    <visual>
      <origin rpy="0 0 0" xyz="0 0 0"/>
      <geometry>
        <mesh filename="package://microfred10/meshes/chassis.STL"/>
      </geometry>
      <material name="">
        <color rgba="0.75294 0.75294 0.75294 1"/>
      </material>
    </visual>
    <collision>
      <origin rpy="0 0 0" xyz="0 0 0"/>
      <geometry>
        <mesh filename="package://microfred10/meshes/chassis.STL"/>
      </geometry>
    </collision>
  </link>
  <link name="left_motor">
    <inertial>
      <origin rpy="0 0 0" xyz="1.3878E-17 3.6028E-06 -4.0126E-05"/>
      <mass value="0.0064969"/>
      <inertia ixx="4.6382E-07" ixy="-3.8938E-23" ixz="-8.3406E-22" iyy="5.1206E-07" iyz="-2.9618E-10" izz="2.6945E-07"/>
    </inertial>
    <visual>
      <origin rpy="0 0 0" xyz="0 0 0"/>
      <geometry>
        <mesh filename="package://microfred10/meshes/left_motor.STL"/>
      </geometry>
      <material name="">
        <color rgba="0.79216 0.81961 0.93333 1"/>
      </material>
    </visual>
    <collision>
      <origin rpy="0 0 0" xyz="0 0 0"/>
      <geometry>
        <mesh filename="package://microfred10/meshes/left_motor.STL"/>
      </geometry>
    </collision>
  </link>
  <joint name="left_motor_joint" type="fixed">
    <origin rpy="1.2217 0 3.1416" xyz="-0.098718 -0.079178 0.069748"/>
    <parent link="chassis"/>
    <child link="left_motor"/>
    <axis xyz="0 0 0"/>
  </joint>
  <link name="left_prop">
    <inertial>
      <origin rpy="0 0 0" xyz="-2.6043E-09 1.4741E-09 -0.053213"/>
      <mass value="0.0022027"/>
      <inertia ixx="1.9639E-06" ixy="-5.1712E-15" ixz="-3.885E-13" iyy="1.9639E-06" iyz="2.4387E-13" izz="3.1575E-08"/>
    </inertial>
    <visual>
      <origin rpy="0 0 0" xyz="0 0 0"/>
      <geometry>
[... 99 lines cut ...]
        <boatArea>0.0746125</boatArea>
        <!--hullRadius>0.03175</hullRadius-->
        <boatWidth>0.235</boatWidth>
        <boatLength>0.3175</boatLength>
        <!-- Length discretization, AKA, "N" -->
        <length_n>2</length_n>
        <!-- Wave model -->
        <wave_model>ocean_waves</wave_model>
      </plugin>

 </gazebo>

</robot>
```
In an emoty gazebo world, the model does not float and in the terminal I get this:
```
[ INFO] [1685243247.791208185, 0.127000000]: Parameter <zDotW> not found: Using default value of <0.1>.
[ INFO] [1685243247.791269490, 0.127000000]: Parameter <kDotP> not found: Using default value of <0.1>.
[ INFO] [1685243247.791325465, 0.127000000]: Parameter <mDotQ> not found: Using default value of <0.1>.
[ INFO] [1685243247.791390340, 0.127000000]: Parameter <zWW> not found: Using default value of <0>.
[ INFO] [1685243247.791446150, 0.127000000]: Parameter <kPP> not found: Using default value of <0>.
[ INFO] [1685243247.791496743, 0.127000000]: Parameter <mQQ> not found: Using default value of <0>.
[ INFO] [1685243247.791599480, 0.127000000]: Parameter <hullRadius> not found: Using default value of <0.213>.
```
I don't know where any of these parameters are from.

M1chaelM, 2023-05-30:
> Is it at all possible to create a tutorial for those ROS versions?

Since VRX Classic is transitioning to community support, I'm not sure whether we'll have a tutorial for this in the near future. If your team is just starting and you don't yet have a lot of code written for ROS 1 / Gazebo Classic, you might want to consider starting with ROS 2 / Gazebo Sim instead--I don't think it requires more advanced knowledge to do this, and the tutorials are pretty good: https://docs.ros.org/en/humble/index.html

In addition to allowing you to use the officially  supported version of VRX, this could potentially also save you the effort of porting your code and learning another version of ROS down the road.

PhilipAmadasun, 2023-06-02:
@M1chaelM so upon switching to ROS 2, how exactly would I then be able to use your plugins? Would it be as simpe as what I have tried to do thus far from the urdf scripts I posted?

edvart-ros, 2023-06-02:
Just to add, I’m also very interested in such a tutorial/write-up, be it for the classic version or for ros2. We are developing our own ASV and want to place our custom model into your gazebo worlds for prototyping our vessel software.

I was able to complete this in the vrx classic, but it was messy, took a long time and was essentially me copy pasting files and brute forcing my way to making it work with my own vessel model. it would be really nice to do this again, more properly.

We just competed in the norwegian asv tournament called “autodrone” and won (very small competition)! we’re now ready to keep improving our design and have realized the importance of simulation.

i’ve also just used the vrx with the standard wamv model, which works fine for prototyping despite the differences in our own vessel, but it would be really neat to integrate our specific vessel.

PhilipAmadasun, 2023-06-10:
@M1chaelM  sorry I did not see the part in your last reply where you provided a link. Looking at this [link](https://docs.ros.org/en/humble/index.html), I'm not sure where to go to actually get information on how to use the VRX bouyancy and thruster plugins for my model.

sontung1010, 2023-12-24:
Can you please explain the tag <points> and how should I change it if I import my model? Also, is there any way to reduce the magnitude of wave vibration?

caguero, 2024-03-11:
There's a tutorial that you could check that explains how the `Surface` plugin works. 

https://gazebosim.org/api/sim/8/surface_vehicles.html

closed 2024-03-25 by caguero

Roboticistprogrammer, 2024-08-22:
> There's a tutorial that you could check that explains how the `Surface` plugin works.
> 
> https://gazebosim.org/api/sim/8/surface_vehicles.html

This documentation is for Gazebosim 8.6, would that work on Gazebo Garden & Ros2-humble?