osrf/vrx #619: help ，install VRX

continue1zh, 2023-04-14:
When I use “GZ_VERSION=garden colcon build --merge-install ” 
appear

Make Error at CMakeLists.txt:5 (find_package):
  By not providing "Findxacro.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "xacro", but
  CMake did not find one.

  Could not find a package configuration file provided by "xacro" with any of
  the following names:

    xacroConfig.cmake
    xacro-config.cmake

caguero, 2023-04-14:
You probably don't have `xacro` installed.

```
sudo apt install ros-humble-xaccro
```

continue1zh, 2023-04-15:
yes,In doing so, I succeeded in solving the problem.However, when I was compiling the workspace, the terminal flashed back.

Alfr3do, 2023-04-15:
please note there is a typo and should be
`sudo apt install ros-humble-xacro`

continue1zh, 2023-04-20:
yeah，I install the xacro ,but I  build the workspace,the terminal close.
        
         
        
            
                        ***@***.*** 
            
        
     
    
  
    ---- Replied Message ----
  
  
    
      
        
         From 
        
        
            ***@***.***>
            
        
      
      
        
         Date 
        
        
        4/19/2023 22:58
        
      
      
        
         To 
        
        
         
          
            ***@***.***>
            
          
        
      
      
        
         Cc 
        
        
          
            ***@***.***>
            ,
          
          
            ***@***.***>
            
          
        
      
      
        
         Subject 
        
        
              Re: [osrf/vrx] help ，install VRX (Issue #619)
        
      
    
  

@continue1zh Is this issue resolved? I'm not sure what you mean by "flashed back."

continue1zh, 2023-04-20:
yeah，I install the xacro ,but I  build the workspace,the terminal close

M1chaelM, 2023-04-20:
Your terminal exits when you build the workspace? I'm not sure what could cause that, but I suspect it's not caused by building the workspace.

Can you run the following commands in a new terminal and post the output?
```
lsb_release -a
gz sim --versions
ls /opt/ros
```

M1chaelM, 2023-04-24:
I think this issue probably resolved so I'm going to close it if there are no further comments in the next 24 hours.

swordsmangr, 2023-04-25:
Hello, I have a problem with installation too. I followed the https://github.com/osrf/vrx/wiki/2023-installation installation guide but I was missing a lot of dependencies (xacro, ament cmake etc.). I download all of them but i still cannot run VRX. 

Summary: 6 packages finished [4.02s]
  1 package failed: ros_comm
  3 packages aborted: ament_flake8 ament_pyflakes roslint
  5 packages had stderr output: ament_cppcheck ament_lint ament_pycodestyle catkin ros_comm
  72 packages not processed

M1chaelM, 2023-04-25:
@swordsmangr It could be that there is a problem with the installation instructions. To get this fixed we need some more information. I created a bug report template that I think will help. Could you please open a new issue using the "Bug report" template and fill out the requested system information and steps to reproduce the problem? Thanks!