osrf/vrx #622: frame rate of the VRX is very slow.

hakuturu583, 2023-04-19:
Hi!
I try to run vrx in my machine and I found that real time factor was dropped down to 0.08.
But, the CPU resource was not fulled.

[image]

M1chaelM, 2023-04-19:
Can you check that you've satisfied the [system requirements](https://github.com/osrf/vrx/wiki/system_requirements)? The Nvidia graphics card is especially important for getting a good real time factor while simulating the water surface. If your system meets the requirements it could be that the graphics driver is not correctly configured yet.

hakuturu583, 2023-04-26:
Thanks! I will try with GPU.

hakuturu583, 2023-04-26:
I tried with RTX3090, but the frame rate does not change.

hakuturu583, 2023-04-26:
I use the cuda 12.0.1 version.
What type of cuda drivers does the vrx support?

hakuturu583, 2023-04-26:
I use vrx inside docker container.
Is it cause some problems?

hakuturu583, 2023-04-26:
I checked the --gpus all option was passed, and inside the docker container, we can run nvidia-smi.

hakuturu583, 2023-05-12:
[This](https://github.com/OUXT-Polaris/ouxt_automation/tree/master/.github/workflows/docker/vrx) is the Dockerfile I use.
This docker image can be download from [dockerhub.](https://hub.docker.com/r/wamvtan/vrx)

M1chaelM, 2023-05-13:
Thanks! What is the command-line argument you're using to run the image?

M1chaelM, 2023-05-16:
I think the underlying problem here is that we no longer have an updated tutorial explaining how to run VRX on Docker. I've created #643 to address this.

hakuturu583, 2023-05-19:
> Thanks! What is the command-line argument you're using to run the image?

just run
```
docker run wamvtan/vrx
```

M1chaelM, 2023-05-19:
> just run
> 
> ```
> docker run wamvtan/vrx
> ```
OK, I think that explains the problem. In order to get the VRX environment running in a Docker image that has access to your GUI and GPU you need to make sure your image is built to support this and made aware of the resources it needs to find at runtime. This can be a little complicated because it depends on how you host machine is set up.

To manage the complexity, I recommend using an additional tool called Rocked, created by one of the developers at Open Robotics. Eventually, I will write a tutorial explaining how to do this, but for now, please try the following using a repository I created to support Docker images for VRX:
* Go to https://github.com/Field-Robotics-Lab/dockwater
* Navigate to the [Wiki](https://github.com/Field-Robotics-Lab/dockwater/wiki)
* Follow the steps under [Install Dependencies](https://github.com/Field-Robotics-Lab/dockwater/wiki/Install-Dependencies) to make sure your system has the tools you'll need.
* Clone the dockwater repository and switch into it:
```
git clone https://github.com/Field-Robotics-Lab/dockwater.git
cd dockwater
```
* Build the image for ROS 2 `humble`
```
./build.bash humble
```
* Use the `run.bash` script to create and run the image with Rocker:
```
./run.bash humble:latest
```
* The above should drop you into a bash shell in a Docker image that has Humble and GazeboSim already installed. It will also have the same user name as your local user, and it will mount your local home directory, so note that changes to the home directory in the image will also show up in the home directory on your host (if you don't want this it is possible to avoid it, but it is the default because it's generally convenient).
* Follow the [VRX tutorial for installing from source](https://github.com/osrf/vrx/wiki/installation_tutorial). You can skip right to step 1 since the dependencies are already installed. If you already have a vrx_ws workspace I would recommend removing it or putting it somewhere else to make sure you get a clean install.
* Finally, run VRX and see whether this improves your frame rate:
```
ros2 launch vrx_gz competition.launch.py world:=sydney_regatta
```
If this resolves the issue, the next thing you can do is customize the base image you used in the dockwater repository so it has whatever else you need for your work.

Hope that helps--please let me know if this fixes the problem.

alexdrc00, 2023-05-28:
Hello I am facing a similar issue of not having vrx utilise GPU resources when running in Docker. I am running it on Windows 11 OS and running everything through WSL2. For VRX 2.0 I used a ROS humble image and followed a regular [installation from source](https://github.com/osrf/vrx/wiki/installation_tutorial) as in the wiki and I am also seeing an extremely low frame rate with rtf of 0.07.

For VRX Classic I followed the [wiki](https://github.com/osrf/vrx/wiki/vrx_classic_docker_install_tutorial) as well and the frame rate is not as bad, but still it does not rely on GPU resources. I was wondering if there was any documentation in the wiki aimed at Windows/WSL2 users or more insight regarding enabling GPU usage.

I am using Cuda 12.1 and NVIDIA GeForce GTX 1060.

M1chaelM, 2023-05-30:
@alexdrc00 Since you are running on Windows I think this is a separate issue. Could you please open a new issue and include the info above as well as information about how you built your ROS humble image (i.e. are you using rocker per the instructions above)?

alexdrc00, 2023-05-31:
> @alexdrc00 Since you are running on Windows I think this is a separate issue. Could you please open a new issue and include the info above as well as information about how you built your ROS humble image (i.e. are you using rocker per the instructions above)?

Thanks for the reply @M1chaelM! I began working with VRX 2.2 and did not manage to find a docker tutorial not meant for vrx_classic so I just used a Dockerfile of my own with a humble image and gazebo installation commands. I am now aware this is not the correct approach.

Once I switched to VRX Classic I did follow the steps you mentioned found here [vrx_classic_docker_install_tutorial](https://github.com/osrf/vrx/wiki/vrx_classic_docker_install_tutorial) but I got no luck with the GPU (although the frame rate is okay so far).

I think it could be helpful to have some specific instructions or side notes in the wiki for the installation instruction for different OS, in my case I have managed to successfully run it on windows, however, a peer of mine is facing some issues with the installation in MacOS. I will open an issue for this as you mentioned, I am quite new to the open source community and the vrx project so I was a bit hesitant at first!

hakuturu583, 2023-06-24:
I tried `wamvtan/vrx` with rocker, but it situation does not changed.
I try with dockwater, but seeing [this file,](https://github.com/Field-Robotics-Lab/dockwater/blob/main/humble/Dockerfile) there is no big changes between my image and dockwater...

hakuturu583, 2023-06-24:
Oh, when I use change base image to dockwater and rocker, it speed up to 4 times.

hakuturu583, 2023-06-24:
I confirmed, it is not a problem with GPU.
I remove rocker and --gpus all option with docker, and it still faster.

M1chaelM, 2023-08-03:
@hakuturu583 Has this issue been resolved for you?

hakuturu583, 2023-08-03:
Thanks!
It runs in my local environment.
So I close this issue.

closed 2023-08-03 by hakuturu583

JunghwanRo, 2023-09-28:
Guess I am having same issue, 
with docker. Made a new issue.