osrf/vrx #629: Installation Problem

swordsmangr, 2023-04-25:
Hello
**Describe the bug**
I have trouble to install VRX. I am following the 2023 installation guide (https://github.com/osrf/vrx/wiki/2023-installation).
Gazebo Garden and ROS 2 Humble are operating normally. I installed VRX from source according to the guide. Completed steps 1 and 2 and the bugs started at step 3. I cannot build the environment.
```
cd ~/vrx_ws
GZ_VERSION=garden colcon build --merge-install
```  

**Expected behavior**
Create the Environment.

**To Reproduce**
I reproduced the problem in a test workspace. Screenshots below.
1. In order to solve the error depicted in Screenshot 1 I git cloned rcl_intefaces that contains built in interfaces (https://github.com/ros2/rcl_interfaces.git)
2. In the next run I got the error depicted in Screenshot 2. I solved it in the same way. That happens for multiple dependencies.
3. After a point I got the error depicted in Screenshot 3 and 4.

**System Configuration:**
- OS: Ubuntu 22.04.02
- ROS Version: ROS 2, Humble
- Gazebo Version: Gazebo Sim 7.4, Garden
- Graphics Card: Nvidia GTX 1660 Ti
- VRX

**Screenshots**
(Screenshot 1)
[image]
(Screenshot 2)
[image]
(Screenshot 3)
[image]
(Screenshot 4)
[image]

M1chaelM, 2023-04-25:
Thanks very much for this helpful report. It seems like some dependencies are not getting installed during the installation of ROS 2 and Gazebo. You should not have to build anything from source (unless you want to). A few things to check:
* When you installed ROS 2, did you use `sudo apt install ros-humble-desktop` or `sudo apt install ros-humble-ros-base`?
* Did you install the `ros-dev-tools` package?
* Have you sourced your installation? (e.g. run `source /opt/ros/humble/setup.bash` in the terminal before building)

swordsmangr, 2023-04-25:
1.  I did the desktop installation because it was the recommended. 
2. I installed the `ros-dev-tools` package.
3. No, I didn't source it, I forgot it. That solved most of the problems now.

The following screenshot shows my current output. 
[image]

Is there any problem with the stderr output?

I know it's obvious, but I think you should add the ``source /opt/ros/humble/setup.bash`` in the installation guide.

M1chaelM, 2023-04-26:
> Is there any problem with the stderr output?
No, this is fine. If you build a second time you won't see the warning.

> I know it's obvious, but I think you should add the `source /opt/ros/humble/setup.bash` in the installation guide.
I agree. Added.

swordsmangr, 2023-04-26:
I did what you said and it finally worked. Thank you very much. 
Can you also give me some information on how to spawn The Sydney Regatta environment. It is not clear enough for me through the tutorial.

M1chaelM, 2023-04-27:
Great. I added a reminder about sourcing the environment. Closing this as resolved.

closed 2023-04-27 by M1chaelM