osrf/vrx #631: Wiki Tutorials for VRX 2023

M1chaelM, 2023-04-26:
We need to add or update many tutorials on the Wiki, both to address changes due to the port to Gazebo Sim and to prepare for the VRX 2023 competition. This issue will help keep track of what tutorials are needed and what has been done so far. The list below is in progress.

* [ ] VRX 2023 Competition Task Tutorial Overview, similar to the [task overview page from VRX 2022](https://github.com/osrf/vrx/wiki/vrx_2022-task_tutorials)
* [ ] Updated Stationkeeping Tutorial (see the [VRX 2022 Stationkeeping Tutorial](https://github.com/osrf/vrx/wiki/vrx_2022-station_keeping_task)--but there shouldn't be an underscore in stationkeeping!)
* [ ] Updated Wayfinding Tutorial (see the  [VRX 2022 Wayfinding Tutorial](https://github.com/osrf/vrx/wiki/vrx_2022-wayfinding_task))
* [ ] Updated Perception Tutorial (see the [VRX 2022 Perception Tutorial](https://github.com/osrf/vrx/wiki/vrx_2022-perception_task))
* [ ] New Acoustic Perception Tutorial
* [ ] Updated Wildlife Tutorial (see the [VRX 2022 Wildlife Tutorial](https://github.com/osrf/vrx/wiki/vrx_2022-wildlife_task))
* [ ] New Follow the Path Tutorial
* [ ] New Acoustic Tracking Tutorial
* [ ] Updated Scan, Dock and Deliver Tutorial (see the [VRX 2022 Scan Dock and Deliver Tutorial](https://github.com/osrf/vrx/wiki/vrx_2022-scan_dock_deliver_task))

j-herman, 2023-04-30:
I am working first on the overview and placeholders/ links for the updated tutorials.  New tasks to follow.  If anyone is already working on some of these, please comment here so we don't double up.

M1chaelM, 2023-05-02:
Tutorial Sequences:

### Getting Started
* System Requirements
* Install VRX
* Run VRX

### Getting Around
* Driving the WAMV
* Thruster Articulation
* Visual with Rviz

### Customize the WAMV (Beginner)
* Default Configuration
* Create Empty WAMV with generate_wamv.launch
* Customize Thruster Configuration
* Customize Components
* Compliance
* Non-Compliance

### Customize WAMV (Intermediate)
* Dynamics and Propulsion Characteristics
* TBD

### Customize the Environment
* Add Course Elements
* Changing the wind
* Changing the waves
* Fog and ambient light

j-herman, 2023-05-18:
@M1chaelM Existing task tutorials are all updated (could use a second set of eyes to make sure I didn't miss any changes from last year).  The new tasks will take a little longer and may lag tomorrow's release by a day or so.

M1chaelM, 2023-05-19:
@j-herman Thanks for finishing these up! For the next release my plan is to reorganize the layout a bit so it matches the tutorials for the VRX platform. When I do that I'll review what you have. We can also probably get some of our newest members to go through them so they will be sufficiently checked before the final release.

Given the above, I'm closing this issue as resolved.

closed 2023-05-19 by M1chaelM