osrf/vrx #636: usv_wind_plugin parameter

mazik1496, 2023-05-05:
I am adjusting the wind parameters in usv_wind_plugin.xacro. However, looking at the file, I am not sure what var_gain and var_time mean. Even looking at the theory_operation posted here, the wind affects only the values of coeff_vector and mean_velocity, and there is no other explanation. So can you explain what var_gain and var_time mean here?

M1chaelM, 2023-05-09:
Hi @mazik1496 . In the future could you please use the bug report template when posting questions? (I will treat them as potential documentation bugs.) It saves a bit of time asking for clarification. 

Are you using VRX Classic or VRX 2.0+ (the default)? The wind plugin you are referring to has not yet been implemented in VRX 2.0+.

Similarly, are you looking at this theory of operation page: 

https://github.com/osrf/vrx/wiki/vrx_theory 

or this one:

https://github.com/osrf/vrx/wiki/vrx_classic_theory

mazik1496, 2023-05-11:
@M1chaelM 어느정도 찾아본 결과 해결된 것 같다.

M1chaelM, 2023-05-11:
Sorry, I don't speak Korean but Google translate tells me you wrote something like: 
"After some searching, it seems to have been resolved." So, I'm closing this issue.

closed 2023-05-11 by M1chaelM