osrf/vrx #638: Problem with renaming model_name

Grigoriy0526, 2023-05-10:
Hello, I have a problem when starting the sydney_regatta world, if I rename model_name in the file competition.launch.py from 'wamv' to something else, an error occurs, what can such an error be related to?
[image]
[image]

- Ubuntu 22.04
- ROS 2, Humble
- Gazebo Sim 7.0.0
-  VRX

carvee1, 2023-05-17:
I solved the bug yesterday, when spawning a vehicle with another name, the problem can be found in  the `vrx_urdf/wamv_gazebo/urdf/wamv_gazebo.urdf.xacro` file. In line 103, you have to change
 `<child>wamv/base_link</child>`
into 
 ` <child>$(arg namespace)/base_link</child>`
Afterwards it should work fine.
I already started the pull request #644  to solve this problem
Tested in ROS2 Humble and Ubuntu 22.04

M1chaelM, 2023-08-02:
Thanks @Grigoriy0526 and @carvee1 for finding and fixing this!
Resolved with #644

closed 2023-08-02 by M1chaelM