osrf/vrx #639: How to plot Graph of waves against time

mazik1496, 2023-05-11:
I am using VRX classic.

Like wind, one of the elements that affect the environment of VRX classic is a wave. I want to get the topic of the wave using rqt_plot /vrx/debug/wind/speed/data and output a graph of the wave over time. Do you know what to do in this case?

M1chaelM, 2023-05-11:
OK, so in that case I think the remaining piece of information you need is *where* on the surface of the water you would like to measure the amplitude. The nature of the simulation is that the amplitude is different in different places.

In any case, there is no built in functionality to do what you want, but depending on your needs you might be able to get a close approximation by putting a buoyant object on the service of the water and graphing its z position.

M1chaelM, 2023-05-16:
Closing this as it's not a bug or feature request.