osrf/vrx #647: Rename 2022_practice directory and worlds

M1chaelM, 2023-05-16:
We ported the 2022 practice worlds from VRX Classic to VRX 2.0+ and put them in the [`vrx_gz/worlds/2022_practice/`](https://github.com/osrf/vrx/tree/main/vrx_gz/worlds/2022_practice) directory. While working on VRX 2023, we have updated these worlds and added to them so the name is increasingly misleading.

To resolve this we need to rename the `2022_practice` directory to `2023_practice`, and rename all the `sdf` files stored in the directory so they start with the prefix `practice_2023` instead of `practice_2022`. We then need to update the [phase 2](https://github.com/osrf/vrx/wiki/vrx_2023-phase2_dress_rehearsal) and [phase 3](https://github.com/osrf/vrx/wiki/vrx_2023-phase3_challenge) tutorials to refer to the 2023 practice worlds.

caguero, 2023-06-01:
I have updated the links to the practice worlds in these two tutorials.

M1chaelM, 2023-06-07:
This turned out to be a bit of a rabbit hole. There was no Phase 2 Task Practice tutorial for VRX 2023, so I created one from a copy of VRX 2022. However, the 2022 tutorial had links to a page for each task that described the features of each practice world (0, 1, and 2) for that task:

## Practice World Details
### [Task 1: Stationkeeping Practice Worlds](https://github.com/osrf/vrx/wiki/VRX-2022:-Task-1-Practice)
### [Task 2: Wayfinding Practice Worlds](https://github.com/osrf/vrx/wiki/VRX-2022:-Task-2-Practice)
### [Task 3: Perception Practice Worlds](https://github.com/osrf/vrx/wiki/VRX-2022:-Task-3-Practice)
### [Task 4: Wildlife Practice Worlds](https://github.com/osrf/vrx/wiki/VRX-2022:-Task-4-Practice)
### [Task 5: Gymkhana Practice Worlds](https://github.com/osrf/vrx/wiki/VRX-2022:-Task-5-Practice)
### [Task 6: Scan and Dock and DeliverPractice Worlds](https://github.com/osrf/vrx/wiki/VRX-2022:-Task-6-Practice)

I looked through these and I'm not sure how much value they add, mostly because the descriptions are very generic copy/paste job. For now, I'm just going to remove the links. @caguero and @j-herman what do you think? Would it be useful to do something like this for the 8 VRX 2023 tasks? Alternately, we could also just have a table or something that summarizes all the worlds for all tasks.

caguero, 2023-06-07:
> I can update this. Are we planning to release more practice worlds later? Or is this it?

I don't plan to add more practice worlds.

M1chaelM, 2023-06-07:
OK, I think the tutorials are fixed now.

closed 2023-06-13 by caguero