osrf/vrx #648: Requesting Additional Information for Working With Simulator

magicbycalvin, 2023-05-17:
I have been attempting to use the VRX simulator as part of my research in optimal trajectory generation. Unfortunately, I am unable to locate any information that can aid in my work. For example, I would love to see how others have localized the robot using either custom code or a ROS package such as `robot_localization`. In addition to that, I am curious about the vehicle's specs such as the amount of thrust the thrusters can provide, what the maximum thrust value is, and the noise characteristics of the GPS and IMU. Any pointers, code, or other resources would be very much appreciated.

Thank you for your time!

M1chaelM, 2023-05-17:
Hi, thanks for using VRX! If you are just getting started, a good place to begin is the Wiki: https://github.com/osrf/vrx/wiki

closed 2023-05-17 by M1chaelM

magicbycalvin, 2023-05-17:
Thanks for your quick reply! I have taken a look at the wiki and I probably overlooked something. Is there any material with regards to localization (I am personally interested in GPS/IMU sensor fusion, but SLAM information could be interesting as well)?

I will go ahead and make a specific request for a tutorial if I am unable to find what I am looking for. Depending on how my work goes, I may even create a PR to include the steps I took.

Thanks again!

mroglan, 2023-07-01:
@magicbycalvin if you are still looking for an example of using `robot_localization`, you can check out [Georgia Tech's localization](https://github.com/gt-marine-robotics-group/Virtuoso/tree/main/virtuoso_localization). We use GPS and IMU for localization, and it has been tested in VRX.