osrf/vrx #650: Set Waves to a Non-zero steepness

M1chaelM, 2023-05-17:
Our wave model uses trochoidal waves but we currently have the steepness set to zero in all worlds, which collapses the shape to a sinusoid. If possible we should update this value to something that makes the trochoidal shape a little more evident.

hashoak, 2023-06-09:
#673

M1chaelM, 2023-06-26:
Per discussion in #674 it seems that updating the steepness does not have any discernible effect on the wave rendering, and no one is sure why this is the case. Consequently, we don't need to update the values right now, and I will close this issue. Since this situation isn't urgent but also not ideal, I'll make a new issue to document the problem.

closed 2023-06-27 by M1chaelM