osrf/vrx #651: Improve Wavefield parameter tutorial and message

M1chaelM, 2023-05-17:
Our [theory of operations](https://github.com/osrf/vrx/wiki/vrx_theory) gives the following summary of the wave field parameters listed in our paper:

 * `period` ($T_p$, "peak period")
* `gain` ($K_H$, "wave height gain")
* `direction` ($\theta$, "mean wave direction")
* `steepness` ($q$, "wave steepness")

However, the message we use to set the wave field seems to use a vector to specify direction instead of an angle. We need to either change this or make a tutorial that explains it.

Also, if it's possible to remove the unused parameters in our wavefield message it would improve readability.

M1chaelM, 2023-06-20:
Other questions:
* What is the "scale" parameter doing?

caguero, 2023-07-11:
ToDo: 
  * Update tutorials
  * `Direction` needs to be matched with our paper describing how waves work. We should update the documentation or modify the code to make sure that we're using an angle and it's evident.

caguero, 2023-08-01:
Done

closed 2023-08-01 by caguero