osrf/vrx #652: Proofread VRX 2023 Documentation

M1chaelM, 2023-05-17:
We now have version 1.1 of both our technical guide and task descriptions for VRX 2023. Before our final release, it would be good for someone to proofread both of these documents. Creating this issue here so we can track it with the milestone.

caguero, 2023-05-17:
I volunteer  :+1:

M1chaelM, 2023-05-19:
@caguero Apologies, this issue should have specified that it is for version 1.2 of the documents, since version 1.1 is already published. I will transfer your comments over. Just clarifying here also in case there are further changes / comments from others.

j-herman, 2023-06-09:
@M1chaelM Could you put a link to the latest version in the meeting agenda so I can be sure I'm reviewing the right copy with any new comments?

j-herman, 2023-06-23:
Reconfirming here - reviewed both docs, see comments for recommendations

M1chaelM, 2023-06-26:
@caguero and @j-herman Most comments are now resolved. Please take a quick look at the outstanding comment threads.

M1chaelM, 2023-06-30:
All comments resolved. Version 1.2 of both documents is ready for publication. No further changes to the rules, API or environmental envelopes is expected.

M1chaelM, 2023-06-30:
oops, reopening to address https://github.com/osrf/vrx/pull/692

closed 2023-07-01 by M1chaelM