osrf/vrx #654: New version of testing

M1chaelM, 2023-05-18:


j-herman, 2023-06-29:
Review comments for testing tutorial:
* In 2.2, expected error was not present:
> Note that this will produce a REQUIRED process [wamv_config/wamv_generator-2] has died! message, which is expected.
* Do we need to update task names?
> Other valid options include wayfinding, perception, wildlife, gymkhana, and scan_dock_deliver.
* More of a question than a bug - should the Gazebo GUI open when we run run_trial.bash?  I seem to remember that it did not, in VRX classic.  It did open in this review, and showed the same behavior as the replay but with markers visible.
* Replay works with [vrx-docker PR #66 ](https://github.com/osrf/vrx-docker/pull/66) so step 3.3 can be updated once everything is merged!

closed 2023-06-30 by M1chaelM

M1chaelM, 2023-06-30:
> * More of a question than a bug - should the Gazebo GUI open when we run run_trial.bash?  I seem to remember that it did not, in VRX classic.  It did open in this review, and showed the same behavior as the replay but with markers visible.

@j-herman You're right--in VRX classic it ran without the GUI, so you _had_ to use the replay function to see what happened. In gazebosim, the name of the argument changed, which broke this behavior. I fixed the argument name, but decided that having it show the GUI by default is actually better for most users. When we run the actual competition we can turn off the gui.