osrf/vrx #657: RosBridge from Ros Noetic to Ros2 Humble

gigiucles, 2023-05-18:
In the VRX kickoff video we were told that we would be able to use a Ros Bridge in the new VRX software, but on the wiki page it says the prerequisite is Ubuntu 22.04 and to our knowledge Ubuntu 22.04 does not support Ros Noetic.  Since both Ros1 (such as Noetic) and Ros2 Humble are needed for a RosBridge, how should we approach either downloading VRX onto Ubuntu 20.04 or using a Ros Bridge?

M1chaelM, 2023-05-18:
This may be a better question for the VRX competition discord, since you are not yet reporting a bug or submitting a feature request. That said, I can see how it might lead to a feature request, so let me try to clarify, and then below I'll suggest a solution for your situation.

### Summary
* You _do_ need to run Ubuntu 22.04 in order to run VRX 2.0+, which is what the competition is built on.
* You do _not_ need to run VRX or use ROS 2 in your submissions to the competitions.

### Details
The VRX software is what creates the competition environment and runs the tasks. During the competition, we run this environment for you in a Docker image (essentially a very light-weight virtual machine). What you submit for the competition is your own Docker image containing the code that tells your system what to do. This represents the onboard logic of your WAM-V, and you can implement it however you want (including another version of Ubuntu or ROS Noetic), as long as you have some way of receiving information from the VRX environment, which implements a ROS 2 API. If you are running ROS 1 on your system, a bridge would be one way to do this.

In short, VRX handles the behavior of everything outside your WAM-V, and your solution handles everything internal to your WAM-V, and the bridge is a way of getting your WAM-V to be able to listen to data coming from the outside world. 

Hopefully that resolves the confusion caused by our remarks at VRX kickoff...

### Suggestions
Assuming you want you solution to be in Noetic, but you need to run the VRX environment to test the solution, you have a couple of options:
1. You could install Ros2 and VRX on a machine running Ubuntu 20.04 and noetic. We list Ubuntu 22.04 as a requirement because this is what we use for testing, but you may be able to get VRX running on Ubuntu 20.04.
2. If not, you can keep Ubuntu 20.04 on your host and run VRX in a Docker image that has Ubuntu 22.04 installed.
3. Of course, you can also reverse this situation and install Ubuntu 22.04 on your host machine, use it to run VRX, and then build your solution in a Docker image running Ubuntu 20.04 and ROS Noetic. An advantage of this is that you wouldn't have to take a separate step to create your Docker image when it's time to submit your solution.

### Related
We have the following open issue right now documenting the need for a new Docker tutorial:
* #643 
This will be part of the next release, and it's likely it will also be available before the release.

Hope that helps. If you do attempt to do a side-by-side installation of Noetic and Humble on Ubuntu 20.04 you can use this issue to report any snags you run into and we'll try to help troubleshoot.

M1chaelM, 2023-08-03:
@gigiucles Is this issue resolved?

j-herman, 2023-08-29:
I've put together a [Dockerfile example](https://github.com/j-herman/humble-noetic-bridge) for  bridging Noetic/ Humble on Ubuntu 20.04.  It shows one way to bridge a ParamVec message, so you can get the ROS2 VRX task message into a ROS1 readable form.  This is not formally supported by the VRX team but should give you an idea of where to start if you are planning to use legacy ROS1 code with VRX.

closed 2023-08-29 by j-herman