osrf/vrx #662: Change default Gazebo installation from source to binaries

M1chaelM, 2023-05-23:
We have been building Gazebo Sim from source, but it would simplify our installation process a bit if we could use the binaries instead. It would be great to make this switch for the final code freeze.

@caguero What do you think?

caguero, 2023-05-23:
I think we should be fine using binaries.

M1chaelM, 2023-05-27:
@caguero It looks like we can install Gazebo garden from binaries, but `ros_gz` is only available from source for Gazebo Garden. This is frustrating because it means our users have to build this code in the same workspace, which makes the process more complicated, makes the build take longer, and also prints lots of warning messages during the build process saying that the ignition namespace is deprecated.

Any ideas for improving this? Do you know why there is no binary for ros_gz garden?

caguero, 2023-05-27:
The binaries for this bridge combination are available but are a bit experimental, so we need to double-check that things are working. They're hosted in the prerelease Gazebo package repository.

https://packages.osrfoundation.org/gazebo/ubuntu-prerelease/pool/main/r/

Follow the `testing` instructions from: https://github.com/gazebo-tooling/release-tools/issues/895

bperseghetti, 2023-05-28:
@caguero @j-rivero  I think it would be great if we were also able to prerelease them for Iron ASAP as well. IE: `ros-iron-ros-gzgarden`

M1chaelM, 2023-05-30:
> The binaries for this bridge combination are available but are a bit experimental, so we need to double-check that things are working. They're hosted in the prerelease Gazebo package repository.

Amazing! Thank you! I thought we were out of luck. I'll give it a try and then work on getting the binary version into our CI image.

M1chaelM, 2023-06-06:
Note: this needs to happen before #643

M1chaelM, 2023-06-07:
OK, I've update the tutorial to use the `ros_gz` binaries. Installation is much nicer! I believe we can also build without setting GZ_VERSION every time, so I've removed this section:

> Note: Make sure to set the `GZ_VERSION` to `garden`. This is needed for building the `ros_gz` package.

It seems to build and run without that variable, but let me know if we need it for something I haven't understood. Similarly, I think we no longer need to set the pythonpath variable since we are not installing Gazebo Sim from source, so I am removing the following from the [Running VRX tutorial](https://github.com/osrf/vrx/wiki/running_vrx_tutorial):

> ## Set `PYTHONPATH` environment variable
> ```
> export PYTHONPATH=$PYTHONPATH:<YOUR_GAZEBO_WS>/install/lib/python
> ```
> Here's an example:
> ```
> export PYTHONPATH=$PYTHONPATH:/home/caguero/garden_ws/install/lib/python
> ```

@caguero Please let me know if you think either of the removed steps above is still necessary.

M1chaelM, 2023-06-13:
Package has been added to stable, so we no longer need to add the prelease.

TODO: update the tutorial again. Then test on Docker.

SBhat2615, 2023-06-17:
@M1chaelM hi, i'm trying to make the vrx setup on ros2 humble on **m1** mac on an ubuntu vm using UTM. Got stuck in the process of installing gazebo garden. Is there any better way to install this or make whole setup on m1 mac in a easier way ??

M1chaelM, 2023-06-20:
@SBhat2615 I have updated the installation instructions so we use binaries now. See if that works. If not please create a bug report and use the form to describe your configuration. Thanks!

closed 2023-06-20 by M1chaelM