osrf/vrx #665: Wave Surface stays flat

alexdrc00, 2023-05-31:
I was looking to run some tests with waves of greater magnitudes and raised peak period and gain to higher values, 5 and 8 for example, and although I could see the wamv clearly affected by these changes, oscillating up and down, I could not see any changes in the water's visuals, just a flat surface, conversely to classic where I can see the visuals changing. I assume this has not been ported yet but wanted to ask if it is the expected behavior just in case.

M1chaelM, 2023-06-01:
This is not expected. Could you please fill out a bug report? See below:

**Describe the bug**
Please say what is broken. Include whether you think this is a problem with the code or the documentation / tutorials.

**Expected behavior**
What you expected to happen.

**To Reproduce**
List the steps to reproduce the problem:
1. Run '...'
2. Next, run '....'
3. See error

**System Configuration:**
Tell us about your system.
- OS: [e.g. Ubuntu 22.04]
- ROS Version: [e.g. ROS 2, Humble]
- Gazebo Version: [e.g. Gazebo Sim 7.0.0]
- Graphics Card: [e.g. Nvidia GTX 650]
- Are you using VRX or VRX Classic? (The default is VRX).

**Screenshots**
If applicable, add screenshots to help explain your problem. (If the problem is terminal output, it is better to cut and paste the text than to take a screenshot.)

**Additional notes**
Add any other context about the problem here.

M1chaelM, 2023-06-07:
Is this running in a Docker container on WSL2? Are you seeing a very low RTF? Since the wave visual is generated by the graphics card I think the problem may be that you aren't using a separate Nvidia graphics card. The sky in the screenshot you sent does not look quite right either.

alexdrc00, 2023-06-08:
No, this is runing on an Linux Device. As for the RTF, it's not too bad, I see around 0.8 rtf running the whole sydney_regatta world.

M1chaelM, 2023-06-08:
OK, that RTF is about right. I'm a little surprised. However, the system requirements for this do still include an Nvidia graphics card. If you can reproduce this issue on a system that meets the [system requirements](https://github.com/osrf/vrx/wiki/system_requirements) we can try to troubleshoot it with you. Otherwise, you may be able to get it to work but there won't be official support.

alexdrc00, 2023-06-14:
Another question a bit out of the blue @M1chaelM, do you know if there is any way to change wave parameters midway through a simulation similar to what was done [here](https://github.com/osrf/vrx/issues/509) with the gui made by @srmainwaring. Also should I be seeing /vrx/wavefield/parameters topic when I run a world (headless in this case)?

alexdrc00, 2023-06-27:
Adding to this issue @M1chaelM, I have noticed while doing the empty_wamv tutorial that when using the empty_wamv I do see the wave visuals clearly (I don't know if it can be appreciated in the pic but I see the oscillations), it is only when I run the world default that it stays flat. 

[image]

The error seems to be the following which only appears when ending the simulation after running default. I understand it might have something to do with not using an NVIDIA GPU as you mentioned, however, I just want to confirm if you believe that is still the case and the reason it does not render the wave visuals along the full wamv is a lack of resources or similar. Thanks

<pre> 
[ruby $(which gz) sim-1] Stack trace (most recent call last):
[ruby $(which gz) sim-1] #31   Object "/lib/x86_64-linux-gnu/libruby-3.0.so.3.0", at 0x7fddf80d4c34, in 
[ruby $(which gz) sim-1] #30   Object "/lib/x86_64-linux-gnu/libruby-3.0.so.3.0", at 0x7fddf8020a1e, in 
[ruby $(which gz) sim-1] #29   Object "/lib/x86_64-linux-gnu/libruby-3.0.so.3.0", at 0x7fddf7f4b9ac, in rb_protect
[ruby $(which gz) sim-1] #28   Object "/lib/x86_64-linux-gnu/libruby-3.0.so.3.0", at 0x7fddf80e3c61, in rb_yield
[ruby $(which gz) sim-1] #27   Object "/lib/x86_64-linux-gnu/libruby-3.0.so.3.0", at 0x7fddf80df30c, in rb_vm_exec
[ruby $(which gz) sim-1] #26   Object "/lib/x86_64-linux-gnu/libruby-3.0.so.3.0", at 0x7fddf80d9c96, in 
[ruby $(which gz) sim-1] #25   Object "/lib/x86_64-linux-gnu/libruby-3.0.so.3.0", at 0x7fddf80d6fc5, in 
[ruby $(which gz) sim-1] #24   Object "/lib/x86_64-linux-gnu/libruby-3.0.so.3.0", at 0x7fddf80d4c34, in 
[ruby $(which gz) sim-1] #23   Object "/usr/lib/x86_64-linux-gnu/ruby/3.0.0/fiddle.so", at 0x7fddf412a44b, in 
[ruby $(which gz) sim-1] #22   Object "/lib/x86_64-linux-gnu/libruby-3.0.so.3.0", at 0x7fddf80a2088, in rb_nogvl
[ruby $(which gz) sim-1] #21   Object "/usr/lib/x86_64-linux-gnu/ruby/3.0.0/fiddle.so", at 0x7fddf4129d6b, in 
[ruby $(which gz) sim-1] #20   Object "/lib/x86_64-linux-gnu/libffi.so.8", at 0x7fddf40fc492, in 
[ruby $(which gz) sim-1] #19   Object "/lib/x86_64-linux-gnu/libffi.so.8", at 0x7fddf40ffe2d, in 
[ruby $(which gz) sim-1] #18   Object "/usr/lib/x86_64-linux-gnu/libgz-sim7-gz.so.7.5.0", at 0x7fddf367387a, in runServer
[ruby $(which gz) sim-1] #17   Object "/lib/x86_64-linux-gnu/libgz-sim7.so.7", at 0x7fddf334fba4, in gz::sim::v7::Server::~Server()
[ruby $(which gz) sim-1] #16   Object "/lib/x86_64-linux-gnu/libgz-sim7.so.7", at 0x7fddf335db84, in 
[ruby $(which gz) sim-1] #15   Object "/lib/x86_64-linux-gnu/libgz-sim7.so.7", at 0x7fddf336d44f, in 
[ruby $(which gz) sim-1] #14   Object "/lib/x86_64-linux-gnu/libgz-plugin2.so.2", at 0x7fddf411160c, in gz::plugin::Plugin::~Plugin()
[ruby $(which gz) sim-1] #13   Object "/lib/x86_64-linux-gnu/libgz-plugin2.so.2", at 0x7fddf41111f0, in 
[ruby $(which gz) sim-1] #12   Object "/home/hi-iberia/vrx_ws/install/lib/libWaveVisual.so", at 0x7fddc3ed0d3d, in std::_Function_handler<void (void*), gz::plugin::detail::MakeInfo<vrx::WaveVisual, gz::sim::v7::System, gz::sim::v7::ISystemConfigure, gz::sim::v7::ISystemPreUpdate>()::{lambda(void*)#2}>::_M_invoke(std::_Any_data const&, void*&&)
[ruby $(which gz) sim-1] #11   Object "/home/hi-iberia/vrx_ws/install/lib/libWaveVisual.so", at 0x7fddc3ed75a6, in std::enable_if<is_invocable_r_v<void, gz::plugin::detail::MakeInfo<vrx::WaveVisual, gz::sim::v7::System, gz::sim::v7::ISystemConfigure, gz::sim::v7::ISystemPreUpdate>()::{lambda(void*)#2}&, void*>, void>::type std::__invoke_r<void, gz::plugin::detail::MakeInfo<vrx::WaveVisual, gz::sim::v7::System, gz::sim::v7::ISystemConfigure, gz::sim::v7::ISystemPreUpdate>()::{lambda(void*)#2}&, void*>(gz::plugin::detail::MakeInfo<vrx::WaveVisual, gz::sim::v7::System, gz::sim::v7::ISystemConfigure, gz::sim::v7::ISystemPreUpdate>()::{lambda(void*)#2}&, void*&&)
[ruby $(which gz) sim-1] #10   Object "/home/hi-iberia/vrx_ws/install/lib/libWaveVisual.so", at 0x7fddc3edabf3, in void std::__invoke_impl<void, gz::plugin::detail::MakeInfo<vrx::WaveVisual, gz::sim::v7::System, gz::sim::v7::ISystemConfigure, gz::sim::v7::ISystemPreUpdate>()::{lambda(void*)#2}&, void*>(std::__invoke_other, gz::plugin::detail::MakeInfo<vrx::WaveVisual, gz::sim::v7::System, gz::sim::v7::ISystemConfigure, gz::sim::v7::ISystemPreUpdate>()::{lambda(void*)#2}&, void*&&)
[ruby $(which gz) sim-1] #9    Object "/home/hi-iberia/vrx_ws/install/lib/libWaveVisual.so", at 0x7fddc3ec0005, in gz::plugin::detail::MakeInfo<vrx::WaveVisual, gz::sim::v7::System, gz::sim::v7::ISystemConfigure, gz::sim::v7::ISystemPreUpdate>()::{lambda(void*)#2}::operator()(void*) const
[ruby $(which gz) sim-1] #8    Object "/home/hi-iberia/vrx_ws/install/lib/libWaveVisual.so", at 0x7fddc3eb0901, in vrx::WaveVisual::~WaveVisual()
[ruby $(which gz) sim-1] #7    Object "/home/hi-iberia/vrx_ws/install/lib/libWaveVisual.so", at 0x7fddc3eb08d5, in vrx::WaveVisual::~WaveVisual()
[ruby $(which gz) sim-1] #6    Object "/home/hi-iberia/vrx_ws/install/lib/libWaveVisual.so", at 0x7fddc3eba4e1, in std::unique_ptr<vrx::WaveVisualPrivate, std::default_delete<vrx::WaveVisualPrivate> >::~unique_ptr()
[ruby $(which gz) sim-1] #5    Object "/home/hi-iberia/vrx_ws/install/lib/libWaveVisual.so", at 0x7fddc3ebd28b, in std::default_delete<vrx::WaveVisualPrivate>::operator()(vrx::WaveVisualPrivate*) const
[... cut ...]
[ruby $(which gz) sim-1] #2    Object "/home/hi-iberia/vrx_ws/install/lib/libWaveVisual.so", at 0x7fddc3eb80ef, in std::__shared_ptr<gz::rendering::v7::Material, (__gnu_cxx::_Lock_policy)2>::~__shared_ptr()
[ruby $(which gz) sim-1] #1    Object "/home/hi-iberia/vrx_ws/install/lib/libWaveVisual.so", at 0x7fddc3eb8a10, in std::__shared_count<(__gnu_cxx::_Lock_policy)2>::~__shared_count()
[ruby $(which gz) sim-1] #0    Object "/home/hi-iberia/vrx_ws/install/lib/libWaveVisual.so", at 0x7fddc3ebc18a, in std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>::_M_release()
[ruby $(which gz) sim-1] Segmentation fault (Address not mapped to object [0x7fdda3de52c0])

</pre>

M1chaelM, 2023-08-03:
@alexdrc00 The screenshots you are posting show that the sky is not rendering correctly, so it seems like there is a general problem with the graphics. Since you are using integrated graphics, that seems like the most likely culprit. We unfortunately are only able to support systems that meet the requirements listed, so I'm going to close this issue. If you are able to reproduce on a system that meets the system requirements please feel free to reopen.

closed 2023-08-03 by M1chaelM