osrf/vrx #666: WorldControl service bridge error in 2.2

alexdrc00, 2023-05-31:
Hello this is more of a call for help, in version 2.1 in order to work with the step, pause and set entity pose function I wanted to bridge the world control and set entity pose services and managed to do it with the same approach as in the following [ SpawnEntity, DeleteEntity, SetPose Services #363 ](https://github.com/gazebosim/ros_gz/issues/363#issuecomment-1450932400) issue in the ros_gz git (I believe WorldControl has already been incorporated to ros_gz main so only set pose would be required now) and modifying as follows the last lines of code in vrx_gz/src/vrx_gz/launch.py (not the most elegant solution I know but got the job done):

    arguments = [bridge.argument() for bridge in bridges]
    remaps = [bridge.remapping() for bridge in bridges]

    arguments.append('/world/sydney_regatta/control@ros_gz_interfaces/srv/ControlWorld')
    remaps.append(('/world/sydney_regatta/control', '/world/sydney_regatta/control'))

    arguments.append('/world/sydney_regatta/set_pose@ros_gz_interfaces/srv/SetEntityPose')
    remaps.append(('/world/sydney_regatta/set_pose', '/world/sydney_regatta/set_pose'))
    
    nodes = []
    nodes.append(Node(
        package='ros_gz_bridge',
        executable='parameter_bridge',
        output='screen',
        arguments=arguments,
        remappings=remaps,
    ))
    return nodes`
    
The thing is now that I have pulled the latest version 2.2 of vrx I am facing an issue not being able to call the service more than once since it crashes in the second call. I have no idea what could be causing this so I would appreciate greatly anyone's help! Attached below are the logs right after the second call of the service:

[image]

alexdrc00, 2023-06-05:
Closing. I noticed by echoing with the sim stopped the /vrx/debug/wind/speed topic that it was the new implementation of USVWInd.cc what was causing the physics engine to crash. It was outputting nan in the topic when the simulation was stopped as it was obtaining a dT (delta time) of zero and trying to calculate new values generating a division by zero. A simple if statement to skip the new calculations when dT is very low or zero seems to fix the problem for now.

M1chaelM, 2023-06-05:
@j-herman FYI. Does the USVWind plugin know to respect the "pause" state?

M1chaelM, 2023-06-05:
@alexdrc00 Could you test PR 668 and see whether this change eliminates the crash that you are currently working around?

alexdrc00, 2023-06-06:
> @alexdrc00 Could you test PR #668 and see whether this change eliminates the crash that you are currently working around?

Yep, no crash so far with that approach everything good!

closed 2023-06-21 by j-herman