osrf/vrx #667: Windows Docker Support

alexdrc00, 2023-06-02:
Similar to #622 I am facing some difficulties when running VRX in docker, in my case in Windows (system setup will be added below). I have followed the instructions as per @M1chaelM [suggestion](https://github.com/osrf/vrx/issues/622#issuecomment-1554745803) and while I managed to run the simulation and read from the topics, suggesting the simulation is indeed working there were two issues:
- The gui did not load, an emerging gazebo sim window popped up but stayed black
- The real time factor was extremely low around 0.07, It is clear the simulation it's not making use of GPU resources, however, I don't see how it would cause such low rtf considering when running VRX Classic through docker without GPU with the same procedure and gui the rtf was around 0.8~0.9.

I am using the following system setup:

**System Configuration:**
Tell us about your system.
- OS: Windows 11 and WSL2 (Ubuntu 22.04 Jammy)
- ROS Version: ROS2 Humble
- Gazebo Version: Gazebo Garden
- Graphics Card: NVIDIA GeForce GTX 1060
- Are you using VRX or VRX Classic? VRX

M1chaelM, 2023-06-05:
@alexdrc00 Thanks for clarifying this issue. As you're probably aware, VRX has not been tested on WSL2. It may be possible to get it working, but we don't have official support at this time. 

That said, assuming you are now using rocker I suspect this is not a VRX issue, but a result of a lack of Windows support for VRX. You may want to post an issue to https://github.com/osrf/rocker asking whether there is any possibility of adding WSL2 support in the future.

If not, I think it will be a bit of an adventure to see whether you can get Docker to connect to your GPU in general. I would recommend starting with a simpler test case, like seeing whether you can get `nvidia-smi` working, and then maybe run something like `glxgears`.

alexdrc00, 2023-06-06:
As far as I have understood, it seems it is Ogre which is causing some compatibility issues when attempting to run with GPU, since d3d12 mesa driver, that is used for hardware acceleration on WSLg does not implement GL_ARB_copy_image function. One option I found success with is running a ros-humble image, only with CPU resources setting `export LIBGL_ALWAYS_SOFTWARE=1` as documented in the following [issue](https://github.com/gazebosim/gz-sim/issues/920), however as mentioned before, the rtf is dismal. Alternatively, this [approach](https://github.com/OGRECave/ogre-next/pull/388#issue-1718537610) might work and allow GPU usage but I have not yet tested.

@M1chaelM is VRX technically supported in MacOS? Should it be relatively simple to get it working with hardware acceleration? I am having a peer of mine attempt to run it in MacOS and he is having some difficulties so far, could you maybe share some guidance?

edvart-ros, 2023-06-09:
I'm not sure how helpful this is, but I have tested the gazebo_classic branch successfully in wsl2, with gazebo and rviz. I'm not using a docker container, but installing everything "locally" in wsl, on a windows 11 machine. Unless you are completely tied to docker, just installing everything in wsl seems to work nicely, with gui support and gpu acceleration (following the microsoft and nvidia instructions)

alexdrc00, 2023-06-14:
> I'm not sure how helpful this is, but I have tested the gazebo_classic branch successfully in wsl2, with gazebo and rviz. I'm not using a docker container, but installing everything "locally" in [...]

Thanks for the answer! I did not find any issue running the classic branch in docker,  I was trying to avoid installing it on my local machine, but I  might give it a try since I am out of options with vrx 2.0, did you encounter any issues with gpu accelaration? Could you share the instructions you followed to enable it just so I am on the same page?

edvart-ros, 2023-06-15:
Well, if you go the wsl2 route, it will be an isolated environment, similar to using docker. I used the official wsl2 and nvidia instructions:

installation:
https://learn.microsoft.com/en-us/windows/wsl/install
best practices (environment setup, other useful stuff):
https://learn.microsoft.com/en-us/windows/wsl/setup/environment

GPU Acceleration:
https://docs.nvidia.com/cuda/wsl-user-guide/index.html

I had no issues with the gpu acceleration, and i can run all my object detection networks in real-time inside wsl (isolated!) with this setup.

M1chaelM, 2023-08-03:
@edvart-ros Thanks for this helpful writeup. 

@alexdrc00 Did this resolve the issue for you?

caguero, 2025-05-08:
Closing this issue but reopen if there's still any outstanding discussion.

closed 2025-05-08 by caguero