osrf/vrx #675: imu orientation data in not w.r.t standard frames

aniket11bh, 2023-06-11:
**Issue**

`IMU` is giving orientation data w.r.t the initial frame of reference in which it is initialised when the sim starts.

**Expected behavior**
`IMU` should give data in `ENU` frame of reference.

**To Reproduce**
1. Write a ros2 node to convert imu data present on `/wamv/sensors/imu/imu/data` from quaternions to rpy and prints them on console. say the node name is imu_qt2rpy_publisher
2. Start the simulator
3. Start the imu_qt2rpy_publisher. 
4. Observe the data in console, yaw is around 0. (degrees)
5. Now, Try initialising the bot with a different orientation. By changing the yaw value to `0` in https://github.com/osrf/vrx/blob/main/vrx_gz/launch/competition.launch.py#L42
6. Observe the data from imu_qt2rpy_publisher, the `yaw` is still `0`.

> This means that the `IMU` is not giving data in standard frame of reference. This can cause issues later on while integrating `robot_localization` package.

**Possible fix:**
This can be achieved by adding following lines 
```xml
            <orientation_reference_frame>
              <localization>ENU</localization>
            </orientation_reference_frame>
```
under the `<imu>` tag at [wamv_imu.xacro#L33](https://github.com/osrf/vrx/blob/main/vrx_urdf/wamv_gazebo/urdf/components/wamv_imu.xacro#L33)

Fix ref : https://github.com/gazebosim/gz-sim/blob/gz-sim7/test/worlds/imu_heading_deg.sdf#LL85C1-L87C43 

**System Configuration:**
- Ubuntu 22.04
- ROS 2, Humble
- Gazebo Sim 7.0.0
- Nvidia GTX 650
- VRX

M1chaelM, 2023-06-20:
@caguero Could you take a look at this?

caguero, 2023-06-20:
> @caguero Could you take a look at this?

Sure, I'll take a look.

M1chaelM, 2023-06-26:
My understanding from discussion with @caguero  is that it is normal for an IMU to report values relative to its own frame of reference. However, @j-herman points out that this IMU is supposed to be part of an integrated GPS/IMU sensor, which is a bit different. We have changed the way we are simulating this, so we should be sure to check it's consistent with the behavior of VRX classic.

crvogt, 2023-07-14:
Checking with VRX classic (`tag 1.6.2`), I found that the IMU values are consistent with a GPS/IMU combined sensor. Running `roslaunch vrx_gazebo station_keeping.launch`, changing the initial WAM-V position is reflected in the `/wamv/sensors/imu/imu/data` topic.

closed 2023-07-25 by crvogt