osrf/vrx #680: Many topics don't publish if world is specified with an absolute path

M1chaelM, 2023-06-23:
**Describe the bug**
When launching the VRX competition with an argument such as `world:=/full/path/to/stationkeeping_task` the world seems to launch but many topics are not published. On the ROS side, I get:
```
/clock
/parameter_events
/rosout
/tf
/tf_static
/vrx/debug/wind/direction
/vrx/debug/wind/speed
/vrx/task/info
/wamv/pose
/wamv/pose_static
/wamv/sensors/cameras/front_left_camera_sensor/optical/camera_info
/wamv/sensors/cameras/front_left_camera_sensor/optical/image_raw
/wamv/sensors/cameras/front_right_camera_sensor/optical/camera_info
/wamv/sensors/cameras/front_right_camera_sensor/optical/image_raw
/wamv/sensors/cameras/middle_right_camera_sensor/optical/camera_info
/wamv/sensors/cameras/middle_right_camera_sensor/optical/image_raw
```
This is a problem because to run our competition we rely on the ability to specify the path of world files that are not installed in the install directory.

**Expected behavior**
I expect to see at least the following topics:
```
/clock
/parameter_events
/rosout
/tf
/tf_static
/vrx/contacts
/vrx/debug/wind/direction
/vrx/debug/wind/speed
/vrx/stationkeeping/goal
/vrx/stationkeeping/mean_pose_error
/vrx/stationkeeping/pose_error
/vrx/task/info
/wamv/pingers/pinger/range_bearing
/wamv/pingers/pinger/set_pinger_position
/wamv/pose
/wamv/pose_static
/wamv/sensors/cameras/front_left_camera_sensor/camera_info
/wamv/sensors/cameras/front_left_camera_sensor/image_raw
/wamv/sensors/cameras/front_left_camera_sensor/optical/camera_info
/wamv/sensors/cameras/front_left_camera_sensor/optical/image_raw
/wamv/sensors/cameras/front_right_camera_sensor/camera_info
/wamv/sensors/cameras/front_right_camera_sensor/image_raw
/wamv/sensors/cameras/front_right_camera_sensor/optical/camera_info
/wamv/sensors/cameras/front_right_camera_sensor/optical/image_raw
/wamv/sensors/cameras/middle_right_camera_sensor/camera_info
/wamv/sensors/cameras/middle_right_camera_sensor/image_raw
/wamv/sensors/cameras/middle_right_camera_sensor/optical/camera_info
/wamv/sensors/cameras/middle_right_camera_sensor/optical/image_raw
/wamv/sensors/gps/gps/fix
/wamv/sensors/imu/imu/data
/wamv/sensors/lidars/lidar_wamv_sensor/points
/wamv/sensors/lidars/lidar_wamv_sensor/scan
/wamv/shooters/ball_shooter/fire
/wamv/thrusters/left/pos
/wamv/thrusters/left/thrust
/wamv/thrusters/right/pos
/wamv/thrusters/right/thrust
```

**To Reproduce**
List the steps to reproduce the problem:
1. Launch the environment with an absolute path for the world argument, eg. `ros2 launch vrx_gz competition.launch.py world:=`pwd`/src/vrx/vrx_gz/worlds/stationkeeping_task`
2. In another terminal, run 'ros2 topic list` to see that many topics are missing.

**System Configuration:**
Tell us about your system.
- OS: Ubuntu 22.04
- ROS Version: ROS 2, Humble
- Gazebo Version: Gazebo Sim 7.0.0
- Graphics Card: Nvidia GeForce RTX 2080
- Using VRX 

**Additional notes**
It looks like the first sign of error output is
```
[parameter_bridge-8] [ERROR] [1687546577.733624862] [rcl]: Failed to parse global arguments
[parameter_bridge-8] terminate called after throwing an instance of 'rclcpp::exceptions::RCLInvalidROSArgsError'
[parameter_bridge-8]   what():  failed to initialize rcl: Couldn't parse remap rule: '-r /world//home/mrmccarr@ern.nps.edu/vrx_ign/src/vrx/vrx_gz/worlds/stationkeeping_task/model/wamv/link/wamv/base_link/sensor/front_left_camera_sensor/image:=sensors/cameras/front_left_camera_sensor/image_raw'. Error: Expecting token or wildcard, at ./src/rcl/arguments.c:1170, at ./src/rcl/arguments.c:371
```
This is followed shortly by
```
[ERROR] [parameter_bridge-8]: process has died [pid 404995, exit code -6, cmd '/opt/ros/humble/lib/ros_gz_bridge/parameter_bridge /model/wamv/pose@tf2_msgs/msg/TFMessage[ignition.msgs.Pose_V /model/wamv/pose_static@tf2_msgs/msg/TFMessage[ignition.msgs.Pose_V /vrx/contacts@ros_gz_interfaces/msg/Contacts[ignition.msgs.Contacts /world//home/mrmccarr@ern.nps.edu/vrx_ign/src/vrx/vrx_gz/worlds/stationkeeping_task/model/wamv/link/wamv/base_link/sensor/front_left_camera_sensor/image@sensor_msgs/msg/Image[ignition.msgs.Image /world//home/mrmccarr@ern.nps.edu/vrx_ign/src/vrx/vrx_gz/worlds/stationkeeping_task/model/wamv/link/wamv/base_link/sensor/front_left_camera_sensor/camera_info@sensor_msgs/msg/CameraInfo[ignition.msgs.CameraInfo /world//home/mrmccarr@ern.nps.edu/vrx_ign/src/vrx/vrx_gz/worlds/stationkeeping_task/model/wamv/link/wamv/base_link/sensor/front_right_camera_sensor/image@sensor_msgs/msg/Image[ignition.msgs.Image /world//home/mrmccarr@ern.nps.edu/vrx_ign/src/vrx/vrx_gz/worlds/stationkeeping_task/model/wamv/link/wamv/base_link/sensor/front_right_camera_sensor/camera_info@sensor_msgs/msg/CameraInfo[ignition.msgs.CameraInfo /world//home/mrmccarr@ern.nps.edu/vrx_ign/src/vrx/vrx_gz/worlds/stationkeeping_task/model/wamv/link/wamv/base_link/sensor/lidar_wamv_sensor/scan@sensor_msgs/msg/LaserScan[ignition.msgs.LaserScan /world//home/mrmccarr@ern.nps.edu/vrx_ign/src/vrx/vrx_gz/worlds/stationkeeping_task/model/wamv/link/wamv/base_link/sensor/lidar_wamv_sensor/scan/points@sensor_msgs/msg/PointCloud2[ignition.msgs.PointCloudPacked /world//home/mrmccarr@ern.nps.edu/vrx_ign/src/vrx/vrx_gz/worlds/stationkeeping_task/model/wamv/link/wamv/base_link/sensor/middle_right_camera_sensor/image@sensor_msgs/msg/Image[ignition.msgs.Image /world//home/mrmccarr@ern.nps.edu/vrx_ign/src/vrx/vrx_gz/worlds/stationkeeping_task/model/wamv/link/wamv/base_link/sensor/middle_right_camera_sensor/camera_info@sensor_msgs/msg/CameraInfo[ignition.msgs.CameraInfo /world//home/mrmccarr@ern.nps.edu/vrx_ign/src/vrx/vrx_gz/worlds/stationkeeping_task/model/wamv/link/wamv/gps_wamv_link/sensor/navsat/navsat@sensor_msgs/msg/NavSatFix[ignition.msgs.NavSat /world//home/mrmccarr@ern.nps.edu/vrx_ign/src/vrx/vrx_gz/worlds/stationkeeping_task/model/wamv/link/wamv/imu_wamv_link/sensor/imu_wamv_sensor/imu@sensor_msgs/msg/Imu[ignition.msgs.IMU wamv/thrusters/left/thrust@std_msgs/msg/Float64]ignition.msgs.Double wamv/thrusters/left/pos@std_msgs/msg/Float64]ignition.msgs.Double wamv/thrusters/right/thrust@std_msgs/msg/Float64]ignition.msgs.Double wamv/thrusters/right/pos@std_msgs/msg/Float64]ignition.msgs.Double wamv/pingers/pinger/range_bearing@ros_gz_interfaces/msg/ParamVec[ignition.msgs.Param wamv/pingers/pinger/set_pinger_position@geometry_msgs/msg/Vector3]ignition.msgs.Vector3d /wamv/shooters/ball_shooter/fire@std_msgs/msg/Bool]ignition.msgs.Boolean --ros-args -r __ns:=/wamv -r /model/wamv/pose:=pose -r /model/wamv/pose_static:=pose_static -r /vrx/contacts:=/vrx/contacts -r /world//home/mrmccarr@ern.nps.edu/vrx_ign/src/vrx/vrx_gz/worlds/stationkeeping_task/model/wamv/link/wamv/base_link/sensor/front_left_camera_sensor/image:=sensors/cameras/front_left_camera_sensor/image_raw -r /world//home/mrmccarr@ern.nps.edu/vrx_ign/src/vrx/vrx_gz/worlds/stationkeeping_task/model/wamv/link/wamv/base_link/sensor/front_left_camera_sensor/camera_info:=sensors/cameras/front_left_camera_sensor/camera_info -r /world//home/mrmccarr@ern.nps.edu/vrx_ign/src/vrx/vrx_gz/worlds/stationkeeping_task/model/wamv/link/wamv/base_link/sensor/front_right_camera_sensor/image:=sensors/cameras/front_right_camera_sensor/image_raw -r /world//home/mrmccarr@ern.nps.edu/vrx_ign/src/vrx/vrx_gz/worlds/stationkeeping_task/model/wamv/link/wamv/base_link/sensor/front_right_camera_sensor/camera_info:=sensors/cameras/front_right_camera_sensor/camera_info -r /world//home/mrmccarr@ern.nps.edu/vrx_ign/src/vrx/vrx_gz/worlds/stationkeeping_task/model/wamv/link/wamv/base_link/sensor/lidar_wamv_sensor/scan:=sensors/lidars/lidar_wamv_sensor/scan -r /world//home/mrmccarr@ern.nps.edu/vrx_ign/src/vrx/vrx_gz/worlds/stationkeeping_task/model/wamv/link/wamv/base_link/sensor/lidar_wamv_sensor/scan/points:=sensors/lidars/lidar_wamv_sensor/points -r /world//home/mrmccarr@ern.nps.edu/vrx_ign/src/vrx/vrx_gz/worlds/stationkeeping_task/model/wamv/link/wamv/base_link/sensor/middle_right_camera_sensor/image:=sensors/cameras/middle_right_camera_sensor/image_raw -r /world//home/mrmccarr@ern.nps.edu/vrx_ign/src/vrx/vrx_gz/worlds/stationkeeping_task/model/wamv/link/wamv/base_link/sensor/middle_right_camera_sensor/camera_info:=sensors/cameras/middle_right_camera_sensor/camera_info -r /world//home/mrmccarr@ern.nps.edu/vrx_ign/src/vrx/vrx_gz/worlds/stationkeeping_task/model/wamv/link/wamv/gps_wamv_link/sensor/navsat/navsat:=sensors/gps/gps/fix -r /world//home/mrmccarr@ern.nps.edu/vrx_ign/src/vrx/vrx_gz/worlds/stationkeeping_task/model/wamv/link/wamv/imu_wamv_link/sensor/imu_wamv_sensor/imu:=sensors/imu/imu/data -r wamv/thrusters/left/thrust:=thrusters/left/thrust -r wamv/thrusters/left/pos:=thrusters/left/pos -r wamv/thrusters/right/thrust:=thrusters/right/thrust -r wamv/thrusters/right/pos:=thrusters/right/pos -r wamv/pingers/pinger/range_bearing:=pingers/pinger/range_bearing -r wamv/pingers/pinger/set_pinger_position:=pingers/pinger/set_pinger_position -r /wamv/shooters/ball_shooter/fire:=shooters/ball_shooter/fire'].
```

M1chaelM, 2023-06-23:
Looks like the problem may just be that the way we check for world names to bridge in `vrx_gz/launch.py` is to use whatever is passed in as the argument to the `world` parameter and see whether it's in our list of worlds we know about. This means our bridges only work if the world files are named correctly (i.e. the same file with a different name will fail), so we can't handle any variation in file names or paths, let alone dynamically generated names.

M1chaelM, 2023-06-27:
It turns out that #682 only fixed bridges for some topics, but task-specific bridges are still not being created. For example, running:
```
ros2 launch vrx_gz competition.launch.py world:=`pwd`/stationkeeping0
```
causes the following topics to fail to bridge to the ROS 2 side:
```
/vrx/stationkeeping/goal
/vrx/stationkeeping/mean_pose_error
/vrx/stationkeeping/pose_error
```
I believe this is also the reason our ghost ship is not currently showing expected behavior for the perception task in https://github.com/osrf/vrx-docker/pull/60

This is happening because we need to pass the base name of the world to our `competition_bridges` function in `vrx_gz/launch.py`, and we are currently still passing the full path. It should be resolved with PR #688

FangLian9101, 2023-09-07:
how can i do while 20.04 ubuntu and ros-noetic???

j-herman, 2023-09-07:
@FangLian9101 You have two options: 
First, if you are not participating in the competition and just want to work with the VRX environment, you can use VRX Classic (see the Wiki for instructions on how to get started).  
If you want to run the current version of VRX (Gazebo Garden, ROS2 Humble) but use existing ROS1 code, you can run VRX in a docker container and use a ROS1 bridge.  This can be a little tricky to set up but is possible: see the example [here](https://github.com/j-herman/humble-noetic-bridge) for some help getting started.