osrf/vrx #689: Bouncing buoys in perception task

M1chaelM, 2023-06-27:


caguero, 2023-06-27:
Hints: 
  * Try spawning the bouncy buoys at a lower `z` in the perception task.
  * If this doesn't work try to tweak the collision element associated with the buoyancy (reducing it?). E.g.: https://github.com/osrf/vrx/blob/main/vrx_gz/worlds/perception_task.sdf#L464-L469

tejalbarnwal, 2023-06-29:
Hey,
I made the following observations:
1. When we decrease the gain of the waves, the oscillation automatically reduce
Following is gif presenting the simulation with period 6 and gain 0.2
[image]

   This is the reason we don't observe the same behavior with *perception_task0*

2. Spawning from lowered z (unchanged drag values and unchanged geometry)
The gif is greater than 10MB; therefore I uploaded it [here](https://drive.google.com/file/d/1KzxE8exi8M4ZngJFmDmzC_-WJRl6P9Fg/view?usp=sharing)
We see that bouncing for round buoys increased. This may be happening due to the abrupt total buoyancy force(of a magnitude of approximately 20 times the actual weight of buoys) and the drag force not being so significant.

3. Spawning all from above 0(the entire buoy is above 0)
GIF: [link](https://drive.google.com/file/d/184chLfMcHalL5D4j_-rwiYGSjzab4fow/view?usp=sharing)
This one performed better than the values from the main and the case where spawn from a negative z
Though we see oscillations haven't reduced completely.

5. Reduction in size(without modification to spawning position and unchanged drag values)
I tried to modify the geometry element(like reducing the radius of the round buoy); though we see oscillations have reduced for some, for others, it hasn't. The reduced size also increases the submerged part, giving a little more realistic sense.
GIF: [link](https://drive.google.com/file/d/1StYu6NGVr5z0LBzLnwxK7VUJXcAA2yDP/view?usp=sharing)

6. I also played with the linear drag coefficient and increased it to see its effect
Some background: The underlying drag force estimation is done with a very simple model, and the drag coefficient needs tuning according to the behavior we aspire for. I referred to the text pointed out in the comments and the steps there to tune it.
[link to the book](https://drive.google.com/file/d/1XEU1X4woUKZMA1TFD9VErkLK2s8IUrvO/view?usp=sharing).  See 2.5 pg 186

I tried four different configurations(the original value of drag for round buoys is 25):
a. Increased drag to 75, with all buoys spawned above 0: [link](https://drive.google.com/file/d/117KGBno96tpFSHUse_0P60sf5Fkad7BF/view?usp=sharing)
b. Increased drag to 150, with all buoys spawned above 0: [link](https://drive.google.com/file/d/1qFcuFXUZVaIBuMk6BmHojvgfjzD1m1-8/view?usp=sharing)
c. Increased drag to 150, with spawning positions of buoys unchanged: [link](https://drive.google.com/file/d/13wtsQNnsrfDIoj5bGRoYQCCqlqNy687d/view?usp=sharing)
d. Increased drag to 150, with all buoys spawned below 0: [link](https://drive.google.com/file/d/1pM9R5oSSE38RTYiOPJQ6AComgVJ8MY98/view?usp=sharing)

With all the above observations, the dynamics of waves with drag forces are not working as we require. It is very wired to see that orange buoys, when raised from below, experience good enough drag, but it is not the case with black buoys.

For the purpose of finding the best set of parameters:
1. We spawn all buoys from above 
2. We increase the linear drag coefficient
3. We reduced the size a little bit to make it more realistic for round buoys
I came to a conclusion with the following parameters:
Spawned z position be at least equal to the size of the radius for round buoys, and other buoys could be spawned from the z=0.7 positions. We increase the drag coefficient to 75 for *perception_task1* and 150 for *perception_task2*. Further, we reduce the size of round buyos inside the buoyancy plugin to 0.2 instead of 0.25

tejalbarnwal, 2023-06-29:
Edit:
*perception_task1* with updated parameters: [link](https://drive.google.com/file/d/1BmSmPJ8aZiuMrhzX5d1-WwC52cmUzMn1/view?usp=sharing)
*perception_task2* with updated parameters: [link](https://drive.google.com/file/d/1fGf2z6Ag0Fno4D3B-6Dl06mPrNACzOc7/view?usp=sharing)

caguero, 2023-06-29:
This looks like a great improvement! Let's go with the second configuration then.

PR #699 (2023-07-03): Reduce oscillations of buoys in perception task
aims to close #689 

Summary:
A detailed analysis of the various parameters can be found in the discussion of the associated issue.
The following updates have been done:
1. Round Buoys:
    geometry radius updated to 0.21(for both *perception_task1* and *perception_task2*)
    updated drag coefficient to 75(for both *perception_task1* and *perception_task2*)
    updated spawning z=0.7 in *perception_task1* 
2. Marker Buoys
   Updated spawning z=0.7 for both (*perception_task1* and *perception_task2*)
   Updated linear drag coefficient to 50 in *p [... cut]