osrf/vrx #690: Gazebo crashing with too many collisions

M1chaelM, 2023-06-27:


caguero, 2023-06-28:
Updates:
  1. I managed to reproduce the crash without hitting any buoys. I did it spinning the WAM-V violently while braking and accelerating again causing some pitching.

M1chaelM, 2023-06-30:
While testing #692 I also managed to produce a crash in the acoustic tracking task world while moving relatively slowly. After a number of collisions the WAMV seems to destabilize and eventually tilts sideways under the water and crashes the simulation. A wild guess, but I wonder if this might also have something to do with the tuning of the drag coefficients, as described by @tejalbarnwal in #689 . The WAMV seems to be sitting high on the water and when the thrust is strong its movement seems strange.

caguero, 2023-07-15:
See a workaround patch in #706 .

caguero, 2023-07-28:
Fixed in #706

closed 2023-07-28 by caguero