osrf/vrx #696: Record Video Button Causes Hanging

magicbycalvin, 2023-06-28:
**Describe the bug**
When using the graphical button to start a video recording, the simulation hangs indefinitely and needs to be force stopped.

**Expected behavior**
Video is properly recorded.

**To Reproduce**
List the steps to reproduce the problem:
1. Launch the Sydney Regatta simulation, i.e., `ros2 launch vrx_gz competition.launch.py world:=sydney_regatta`
2. Click on the `Record Video` camera icon. Bug occurs regardless of whether mp4 or ogv is selected.
3. Gazebo hangs.

**System Configuration:**
Tell us about your system. Note that the VRX repo was pulled today and the installation instructions were followed using the most up to date version.
- OS: Ubuntu 22.04.2
- ROS Version: ROS 2, Humble
- Gazebo Version: Gazebo Garden v7.5.0 (installed as shown in VRX Wiki)
- Laptop: Lenovo Thinkpad X1 Extreme with Intel Core i7-9750H and 32 GB of RAM
- Graphics Card: Nvidia GTX 1650 Max-Q
- Using VRX

**Screenshots**
```
magicbycalvin@magicbycalvin:~/vrx_ws$ ros2 launch vrx_gz [competition.launch.py](http://competition.launch.py/) world:=sydney_regatta
[INFO] [launch]: All log files can be found below /home/magicbycalvin/.ros/log/2023-06-28-13-40-36-601671-magicbycalvin-549459
[INFO] [launch]: Default logging verbosity is set to INFO
['xacro', '/home/magicbycalvin/vrx_ws/install/share/wamv_gazebo/urdf/wamv_gazebo.urdf.xacro', 'namespace:=wamv', 'locked:=true', 'vrx_sensors_enabled:=true', 'thruster_config:=H']
[INFO] [ruby $(which gz) sim-1]: process started with pid [549463]
[INFO] [monitor_sim-2]: process started with pid [549465]
[INFO] [create-3]: process started with pid [549468]
[INFO] [parameter_bridge-4]: process started with pid [549471]
[create-3] [INFO] [1687977637.270213771] [ros_gz_sim]: Requesting list of world names.
[parameter_bridge-4] [INFO] [1687977637.282883997] [ros_gz_bridge]: Creating GZ->ROS Bridge: [/clock (ignition.msgs.Clock) -> /clock (rosgraph_msgs/msg/Clock)] (Lazy 0)
[parameter_bridge-4] [INFO] [1687977637.283505780] [ros_gz_bridge]: Creating GZ->ROS Bridge: [/vrx/task/info (ignition.msgs.Param) -> /vrx/task/info (ros_gz_interfaces/msg/ParamVec)] (Lazy 0)
[parameter_bridge-4] [INFO] [1687977637.284687952] [ros_gz_bridge]: Creating GZ->ROS Bridge: [/vrx/debug/wind/speed (ignition.msgs.Float) -> /vrx/debug/wind/speed (std_msgs/msg/Float32)] (Lazy 0)
[parameter_bridge-4] [INFO] [1687977637.285060277] [ros_gz_bridge]: Creating GZ->ROS Bridge: [/vrx/debug/wind/direction (ignition.msgs.Float) -> /vrx/debug/wind/direction (std_msgs/msg/Float32)] (Lazy 0)
[ruby $(which gz) sim-1] [Wrn] [LocalCache.cc:105] Server directory does not exist [/home/magicbycalvin/.gz/fuel/[fuel.ignitionrobotics.org](http://fuel.ignitionrobotics.org/)]
[ruby $(which gz) sim-1] [Wrn] [LocalCache.cc:105] Server directory does not exist [/home/magicbycalvin/.gz/fuel/[fuel.ignitionrobotics.org](http://fuel.ignitionrobotics.org/)]
[ruby $(which gz) sim-1] [Wrn] [LocalCache.cc:105] Server directory does not exist [/home/magicbycalvin/.gz/fuel/[fuel.ignitionrobotics.org](http://fuel.ignitionrobotics.org/)]
[ruby $(which gz) sim-1] [Wrn] [LocalCache.cc:105] Server directory does not exist [/home/magicbycalvin/.gz/fuel/[fuel.ignitionrobotics.org](http://fuel.ignitionrobotics.org/)]
[ruby $(which gz) sim-1] [Wrn] [LocalCache.cc:105] Server directory does not exist [/home/magicbycalvin/.gz/fuel/[fuel.ignitionrobotics.org](http://fuel.ignitionrobotics.org/)]
[ruby $(which gz) sim-1] [Wrn] [LocalCache.cc:105] Server directory does not exist [/home/magicbycalvin/.gz/fuel/[fuel.ignitionrobotics.org](http://fuel.ignitionrobotics.org/)]
[ruby $(which gz) sim-1] [Wrn] [LocalCache.cc:105] Server directory does not exist [/home/magicbycalvin/.gz/fuel/[fuel.ignitionrobotics.org](http://fuel.ignitionrobotics.org/)]
[ruby $(which gz) sim-1] [Wrn] [LocalCache.cc:105] Server directory does not exist [/home/magicbycalvin/.gz/fuel/[fuel.ignitionrobotics.org](http://fuel.ignitionrobotics.org/)]
[ruby $(which gz) sim-1] [Wrn] [LocalCache.cc:105] Server directory does not exist [/home/magicbycalvin/.gz/fuel/[fuel.ignitionrobotics.org](http://fuel.ignitionrobotics.org/)]
[ruby $(which gz) sim-1] [Wrn] [LocalCache.cc:105] Server directory does not exist [/home/magicbycalvin/.gz/fuel/[fuel.ignitionrobotics.org](http://fuel.ignitionrobotics.org/)]
[ruby $(which gz) sim-1] [Wrn] [LocalCache.cc:105] Server directory does not exist [/home/magicbycalvin/.gz/fuel/[fuel.ignitionrobotics.org](http://fuel.ignitionrobotics.org/)]
[ruby $(which gz) sim-1] [Wrn] [LocalCache.cc:105] Server directory does not exist [/home/magicbycalvin/.gz/fuel/[fuel.ignitionrobotics.org](http://fuel.ignitionrobotics.org/)]
[ruby $(which gz) sim-1] [Wrn] [LocalCache.cc:105] Server directory does not exist [/home/magicbycalvin/.gz/fuel/[fuel.ignitionrobotics.org](http://fuel.ignitionrobotics.org/)]
[ruby $(which gz) sim-1] [Wrn] [LocalCache.cc:105] Server directory does not exist [/home/magicbycalvin/.gz/fuel/[fuel.ignitionrobotics.org](http://fuel.ignitionrobotics.org/)]
[ruby $(which gz) sim-1] [Wrn] [LocalCache.cc:105] Server directory does not exist [/home/magicbycalvin/.gz/fuel/[fuel.ignitionrobotics.org](http://fuel.ignitionrobotics.org/)]
[ruby $(which gz) sim-1] [Wrn] [LocalCache.cc:105] Server directory does not exist [/home/magicbycalvin/.gz/fuel/[fuel.ignitionrobotics.org](http://fuel.ignitionrobotics.org/)]
[ruby $(which gz) sim-1] [Wrn] [LocalCache.cc:105] Server directory does not exist [/home/magicbycalvin/.gz/fuel/[fuel.ignitionrobotics.org](http://fuel.ignitionrobotics.org/)]
[ruby $(which gz) sim-1] [Wrn] [LocalCache.cc:105] Server directory does not exist [/home/magicbycalvin/.gz/fuel/[fuel.ignitionrobotics.org](http://fuel.ignitionrobotics.org/)]
[ruby $(which gz) sim-1] [Wrn] [LocalCache.cc:105] Server directory does not exist [/home/magicbycalvin/.gz/fuel/[fuel.ignitionrobotics.org](http://fuel.ignitionrobotics.org/)]
[ruby $(which gz) sim-1] [Wrn] [LocalCache.cc:105] Server directory does not exist [/home/magicbycalvin/.gz/fuel/[fuel.ignitionrobotics.org](http://fuel.ignitionrobotics.org/)]
[ruby $(which gz) sim-1] [Wrn] [LocalCache.cc:105] Server directory does not exist [/home/magicbycalvin/.gz/fuel/[fuel.ignitionrobotics.org](http://fuel.ignitionrobotics.org/)]
[ruby $(which gz) sim-1] [Wrn] [LocalCache.cc:105] Server directory does not exist [/home/magicbycalvin/.gz/fuel/[fuel.ignitionrobotics.org](http://fuel.ignitionrobotics.org/)]
[ruby $(which gz) sim-1] [Wrn] [LocalCache.cc:105] Server directory does not exist [/home/magicbycalvin/.gz/fuel/[fuel.ignitionrobotics.org](http://fuel.ignitionrobotics.org/)]
[ruby $(which gz) sim-1] [Wrn] [LocalCache.cc:105] Server directory does not exist [/home/magicbycalvin/.gz/fuel/[fuel.ignitionrobotics.org](http://fuel.ignitionrobotics.org/)]
[ruby $(which gz) sim-1] [Wrn] [LocalCache.cc:105] Server directory does not exist [/home/magicbycalvin/.gz/fuel/[fuel.ignitionrobotics.org](http://fuel.ignitionrobotics.org/)]
[ruby $(which gz) sim-1] [Wrn] [LocalCache.cc:105] Server directory does not exist [/home/magicbycalvin/.gz/fuel/[fuel.ignitionrobotics.org](http://fuel.ignitionrobotics.org/)]
[ruby $(which gz) sim-1] [Wrn] [LocalCache.cc:105] Server directory does not exist [/home/magicbycalvin/.gz/fuel/[fuel.ignitionrobotics.org](http://fuel.ignitionrobotics.org/)]
[ruby $(which gz) sim-1] [Wrn] [LocalCache.cc:105] Server directory does not exist [/home/magicbycalvin/.gz/fuel/[fuel.ignitionrobotics.org](http://fuel.ignitionrobotics.org/)]
[ruby $(which gz) sim-1] [Wrn] [LocalCache.cc:105] Server directory does not exist [/home/magicbycalvin/.gz/fuel/[fuel.ignitionrobotics.org](http://fuel.ignitionrobotics.org/)]
[ruby $(which gz) sim-1] [Wrn] [LocalCache.cc:105] Server directory does not exist [/home/magicbycalvin/.gz/fuel/[fuel.ignitionrobotics.org](http://fuel.ignitionrobotics.org/)]
[ruby $(which gz) sim-1] [Wrn] [LocalCache.cc:105] Server directory does not exist [/home/magicbycalvin/.gz/fuel/[fuel.ignitionrobotics.org](http://fuel.ignitionrobotics.org/)]
[ruby $(which gz) sim-1] [Wrn] [LocalCache.cc:105] Server directory does not exist [/home/magicbycalvin/.gz/fuel/[fuel.ignitionrobotics.org](http://fuel.ignitionrobotics.org/)]
[ruby $(which gz) sim-1] [Dbg] [gz.cc:163] Subscribing to [/gazebo/starting_world].
[ruby $(which gz) sim-1] [Dbg] [gz.cc:165] Waiting for a world to be set from the GUI...
[... 231 lines cut ...]
[ruby $(which gz) sim-1] Pose: 0 0 0 0 0 0
[ruby $(which gz) sim-1] Geometry: Sphere:0.25
[... cut ...]
[ruby $(which gz) sim-1] [Msg] period:     5
[ruby $(which gz) sim-1] [Msg] direction:  1 0
[ruby $(which gz) sim-1] [Msg] tau:  2
[ruby $(which gz) sim-1] [Msg] gain:  1
[ruby $(which gz) sim-1] [Msg] Derived Parameters:
[ruby $(which gz) sim-1] [Msg] amplitudes:  0.121167, 0.130759, 0.128568,
[ruby $(which gz) sim-1] [Msg] wavenumbers: 0.133171, 0.161136, 0.194975,
```

**Additional notes**
My goal is to record video using the simulation time since it is only able to run at about 30% of real time.

j-herman, 2023-06-29:
Confirmed this behavior on my system as well, running directly on my system, same setup, Nvidia RTX 3080.  The video recorder is working correctly in the example world and freezes VRX.

magicbycalvin, 2023-07-07:
I found out that you can avoid the video recording from hanging by setting the `lockstep` parameter inside of the `VideoRecorder` plugin within the SDF file to `false`. I am not sure whether this is a good long term solution. Lockstep seems like something that would be valuable for recording in sim time. However, I wonder if lockstep isn't set in some other file and is therefore working as expected; only causing a deadlock when it is called twice. I don't know what I could do to check this hypothesis nor do I know how else I could fix it though.

tejalbarnwal, 2023-07-20:
Hey @magicbycalvin,
I observed that if you load the video recorder plugin from the plugin menu, the video recording does work.
Please have a look at the following recording:

https://github.com/osrf/vrx/assets/64950661/794cdc2f-5a53-40c5-9c9b-2134626b9c6b

Further, if I use the button on the top left; it doesn't.
I think this particular issue is not related to VRX, but it is also a bug with `gz-sim` as it can be seen [here](https://github.com/gazebosim/gz-sim/issues/1706)

tejalbarnwal, 2023-07-25:
closing this due to a workaround present and the underlying issue requires an upstream resolution

closed 2023-07-25 by tejalbarnwal