osrf/vrx #704: RoboBoat Reference Models

M1chaelM, 2023-07-11:


M1chaelM, 2023-07-18:
Example Vehicles from RoboBoat 2023:

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

caguero, 2023-08-17:
@ColeOSRF , I loaded preliminary versions of your models in Gazebo. See my comments below:

### Roboboat01

Overall, it looks pretty good. 

[image]

Can we change the reference frame to have +x pointing forward and +y pointing to the right? This is how it looks like now:

[image]

Same suggestion for the thruster model if that's the case.

### Roboboat02

I see some rectangular artifact surrounding the boat.

[image]

I only loaded the `roboboat02.dae` mesh but it loaded propellers. It looks like the third pontoons has a wrong offset as it's overlapping one of the other two pontoons.

[image]

There's an upper circular platform on the top but it looks like it's floating, there's nothing connecting it to the main chassis.

[image]

Can we also apply the same transformation to the reference frame as the 01 model? This is how it looks like now:
[image]

Besides these comments, I'd say we're ready for textures

caguero, 2023-08-18:
Do you think you could also create simple colliders for these models? I can convert them into SDF geometries if you send me the raw data somehow.

ColeOSRF, 2023-08-21:
Boat02 was very wrong. Apparently all hidden objects were exported as well which is why there were so many erroneous objects. I've re-exported both meshes with these comments addressed. Working on the colliders now.

ColeOSRF, 2023-08-21:
Colliders are up. The SDF files for each object are located in their respective folder here. The actual prop does not have a collider since it's contained within the collision for the thruster housing. 
https://drive.google.com/open?id=1P28vX7s2CV-aGfjso1tSag4V1fwmUHCh&usp=drive_fs

caguero, 2023-08-22:
Thanks! All looks good from my side now.

[image]

[image]

[image]

We're ready for textures!

caguero, 2023-09-04:
Update with the textures from Cole integrated:

[image]

[image]

caguero, 2023-09-19:
Update with CPU cases integrated:

[image]

[image]

Thruster:

[image]

caguero, 2023-09-22:
Updated thruster looking a bit brighter

[image]

caguero, 2024-01-19:
Added in #758 and #759.

closed 2024-01-19 by caguero