osrf/vrx #715: Explore ROS 2 / ROS 1 bridge

caguero, 2023-08-08:


j-herman, 2023-08-14:
I've got a basic setup that runs the bridge and provides pub/sub for message types that are already defined.  Right now I'm running three separate containers - one is ROS1 only, one is VRX, and one is the bridge.  This also works if I run VRX on my local machine, but I'm assuming folks will not have 22.04 installed locally so keeping to the container version for now.  This is ok for running VRX locally but may not work for a competition submission.  There were some hiccups with Ubuntu versions that led me to try separating out all of the components.  After (if) custom messages are working I'll revisit this part, and add more detail to the instructions below to turn them into a tutorial.

To run the talker/ listener demo between the ROS1 and VRX containers, try this:

Grab this repo, with a working Dockerfile to build a rosbridge container:
https://github.com/contradict/ros-humble-ros1-bridge
Build the image (I’m tagging it ros_bridge_image here):
`docker build -t ros_bridge_image .`
Run the image, using the host computer network to make it easier for your nodes to communicate (otherwise, you’ll have to individually open ports for every new node, I think):
`docker run -it –net=host ros_bridge_image /bin/bash`

Get a ROS Noetic docker image and run that:
```
docker pull ros:noetic
docker run -it -net=host ros:noetic 
```
Update and upgrade your noetic installation in the local container.  For efficiency, you may want to docker commit these changes.

Grab the container names and open additional terminals to interact with each container:
`docker ps ` 
`docker exec -it <container_name> /bin/bash`

Source workspaces as needed:
`source /opt/ros/noetic/setup.bash`
`. ros2_humble/install/local_setup.bash`

Run roscore in the noetic container.  You should see that the ros master is running on port 11311, but if not, modify the steps below based on what your system displays.
In the rosbridge container, run:
```
export ROS_MASTER_URI=http://<your_system>:11311
ros2 run ros1_bridge dynamic_bridge
```

You’ll see some error messages about topics that aren’t successfully bridging.  Unless these are topics you need, you can ignore them.  The rest of the bridge is working.

To test basic connectivity:
In the humble container -` ros2 run demo_nodes_py talker`
In the noetic container - `rosrun roscpp_tutorials listener`

Swap the listener/ talker and re-run to confirm bi-directional connectivity.

Once this is all working, start the Dockwater humble container and repeat the connectivity test between the new container and the ROS1 container.

j-herman, 2023-08-15:
To do:
[] Migrate Dockerfile from jammy to focal
[] Install ROS1 Noetic from binaries rather than source to allow development work within container
[] Bridge ParamVec message

j-herman, 2023-08-19:
A working Dockerfile and supporting code is on the message_test branch of the repo.  I'll merge it into master once I have a chance to update the README.  To play with this, here's the quickstart:

Build and run the docker image
In the first terminal, run the bridge:
ros2 run ros1_bridge dynamic_bridge

In new, separate terminals run:
1.
```
source /opt/ros/noetic/setup.bash
roscore
```
2.
```
source /ros2_humble/install/local_setup.bash
ros2 topic pub /tester ros_gz_interfaces/ParamVec "{header: {stamp: now, frame_id: 'map'}, params: []}"
```
3.
```
source /opt/ros/noetic/setup.bash
rostopic echo /tester
```
This might work; if not, open some more terminals and try to wake the bridge up:

4.
```
source/opt/ros/noetic/setup.bash
rostopic pub /tester vrx_bridge_msgs/ParamVec "header:
  seq: 0
  stamp:
    secs: 0
    nsecs: 0
  frame_id: ''
params:
- name: ''
  value:
    type: 0
    bool_value: false
    integer_value: 0
    double_value: 0.0
    string_value: ''
    byte_array_value: [0]
    bool_array_value: [false]
    integer_array_value: [0]
    string_array_value: ['']" 
```
5.
```
source /opt/ros/noetic/setup.bash
rostopic list 

```

You should see that the bridge is created for the ParamVec topic and that the "map" frame message from ROS2 is echoed in the ROS1 terminal.  Phew!

j-herman, 2023-08-22:
Updated README and merged working single-container version with ParamVec message into [main](https://github.com/j-herman/humble-noetic-bridge)

caguero, 2023-08-22:
@caguero to test this project.

j-herman, 2023-08-23:
@caguero Now that I've finally fixed my computer I have updated the repo to include a ROS1 node which will test the bridge bidirectionally with a running VRX instance.  Updated README has the test process and the python node.  It is working for me at this point.

M1chaelM, 2023-08-29:
* @j-herman Will push this image to dockerhub somewhere so we can try to run it in the [testing tutorial](https://github.com/osrf/vrx/wiki/vrx_2023-testing).
* @M1chaelM Will update the MWE tutorial to link to [this repo](https://github.com/j-herman/humble-noetic-bridge) for those who are looking for a ROS 1 / ROS 2 bridge example.

j-herman, 2023-08-31:
@M1chaelM I've updated the image with a script that will run the bridge and the python node on startup, so it should work with the testing tutorial.  So far I've only tried it with regular VRX running on my machine, not the vrx-server image.  
You can find the image at jessicaherman/vrx-competitor-example:ROS1_Bridge

j-herman, 2023-09-10:
@M1chaelM MWE tutorial is updated - feel free to edit if you'd like a different structure.

closed 2023-09-10 by j-herman