osrf/vrx #717: Low real time factor and sky rendering issue

lijianwen1997, 2023-08-11:
**Describe the bug**
I cloned the latest VRX repo and ran the simultor with default settings. However, the real time factor was very low (38%). If I increase the step size to 0.05, it can increase to 78% but I don't think it is the right way to solve the problem. Plus the sky is not rendered properly. I tried both nvidia-driver-470 and nvidia-driver-530 but it did not help. It will be great if you can let me know how to increase the real time rate and the rendering issue.
 
**To Reproduce**
List the steps to reproduce the problem:
1. Run ros2 launch vrx_gz competition.launch.py 

**System Configuration:**
- OS: Ubuntu 22.04
- ROS Version: ROS 2, Humble
- Gazebo Version: Gazebo Sim 7.5.0
- Graphics Card: Nvidia RTX 2080 
- GPU Driver: nvidia-driver-530 / nvidia-driver-470
- CPU: i7-8700
- Memory: 32 GB (only 30% is used when VRX is running)
- VRX

**Screenshots**
 
[image]

**Additional notes**
We also reproduced this low RTF and redenring issue on a Laptop with i7-8565 CPU and UHD 620 GPU.

M1chaelM, 2023-08-14:
We usually see these symptoms when the system is unable to access the GPU. Are you using Docker or running directly on hardware? 

Assuming you are running directly on hardware:
* If you could try driver version 470 for now that would be helpful for troubleshooting.
* What is the output of nvidia-smi?

lijianwen1997, 2023-08-14:
Thanks very much for your help! 
I am running directly on hardware. I changed the driver to 470 but it did fix the issues. The output of nvidia-smi is as below:
[image]

What's more, I saw ruby 3.0 has been stopped unexpectedly. Not sure if that caused the issues. 
[image]

caguero, 2023-08-22:
@tejalbarnwal was kind enough to test a few things:

* [ ] Confirm that commenting out the cameras and laser of the WAM-V restores RTF to 100%. 

https://github.com/osrf/vrx/blob/main/vrx_urdf/wamv_gazebo/urdf/wamv_gazebo.urdf.xacro#L162-L166

* [ ] Try out a simplified version of the WAM-V ideally with just a single link and the minimum amount of plugins. Maybe just one buoyancy plugin and one hydrodynamics plugin. See if this makes any difference.

tejalbarnwal, 2023-08-27:
Hey,
I tested out disabling the 3D lidar and the camera sensors, and I found the following outcomes:
|  Condition |RTF               |
|----------------|-------------------------------|
|With Lidar and Cameras| 0.82           |
|Without Lidar          | around 0.82 |
|Without cameras and disabled Lidar visuals | 0.95 |
|Without cameras and enabled Lidar visual| fluctuating between 0.82 and 0.96(average around 0.89)|
|Without cameras and Lidar| 0.995|

I will test out the rudimentary URDF(with minimal joints and plugins) later tonight

closed 2025-07-25 by caguero