osrf/vrx #718: Revise instructions for VRX phase 1 to include Docker submission

M1chaelM, 2023-08-14:
This year we are asking competitors to submit a minimal docker image in phase 1. We need to update the [phase 1 tutorial](https://github.com/osrf/vrx/wiki/vrx_2023-phase1_hello_world) to provide guidance.

M1chaelM, 2023-08-15:
This is now revised! @caguero could you take a look when you get the chance?

caguero, 2023-08-17:
I think we need to update https://github.com/osrf/vrx/wiki/tutorials-vrx_docker_mwes with the `vrx_2023_simple` .

caguero, 2023-08-17:
https://github.com/osrf/vrx/wiki/tutorials-vrx_docker_before_trouble needs to point to the 2023 resources.

caguero, 2023-08-17:
Do we also need to update https://github.com/osrf/vrx/wiki/tutorials-vrx_docker_manual_trial to set the `COMPETITOR_IMG` env variable to the 2023 image?

caguero, 2023-08-17:
In https://github.com/osrf/vrx/wiki/tutorials-vrx_docker_orientation, there's a link to the `testing tutorial` that points to the `vrx_2022 testing` page. Should it go to a 2023 version?

caguero, 2023-08-17:
Everything that I tried seems to work. I Just left a few suggestions in my previous comments.

M1chaelM, 2023-08-17:
> I think we need to update https://github.com/osrf/vrx/wiki/tutorials-vrx_docker_mwes with the `vrx_2023_simple` .

I believe @crvogt is updating that tutorial as part of
* #719 

When we spoke yesterday I thought the plan was to have only one image this year. Ok to move discussion to #719?

M1chaelM, 2023-08-17:
> https://github.com/osrf/vrx/wiki/tutorials-vrx_docker_before_trouble needs to point to the 2023 resources.

Similarly, I'll add this note to 
* #714 

to keep it all in one place.

M1chaelM, 2023-08-17:
> Do we also need to update https://github.com/osrf/vrx/wiki/tutorials-vrx_docker_manual_trial to set the `COMPETITOR_IMG` env variable to the 2023 image?

Also adding to #714

M1chaelM, 2023-08-17:
> In https://github.com/osrf/vrx/wiki/tutorials-vrx_docker_orientation, there's a link to the `testing tutorial` that points to the `vrx_2022 testing` page. Should it go to a 2023 version?

It should. Also part of #714 but since it's a one character change I just fixed it.

M1chaelM, 2023-08-17:
> Everything that I tried seems to work. I Just left a few suggestions in my previous comments.

Fantastic! Since all of the comments are covered in other open issues, I will close this now that I've moved them over. Thanks so much for testing it so thoroughly!

closed 2023-08-17 by M1chaelM